nickcoutsos/keymap-editor #119: Fix: Argless Behaviours in hold-tap

TheSast, 2023-07-01:
Currently, putting a `mod-morph` or `macro` inside a `hold-tap` will result in the editor complaining about a missing parameter (thus preventing it's use). 

Manually modifying the ZMK layout as specified by the [docs](https://zmk.dev/docs/behaviors/hold-tap#using-different-behavior-types-with-hold-taps) by adding a dummy parameter will fix the complaint and will make the `hold-tap` work correctly.

nickcoutsos, 2023-07-01:
It should add placeholders automatically when its able to detect them. Can you provide an example? It's possible some of my experimental support for parameterized macros has unintentionally impacted this.

TheSast, 2023-07-01:
I hope this video can show it well enough:

https://github.com/nickcoutsos/keymap-editor/assets/27977196/922ec05d-40e6-48c8-9ab8-7fa0cea50c93

nickcoutsos, 2023-07-01:
Oof! Ok, thanks for the heads up! I'll try to take a closer look as soon as I have some time to myself this weekend.

nickcoutsos, 2023-07-01:
@TheSast haven't tested extensively but it looks like the issue was strictly in how the "incomplete" binding is detected despite placeholders being set correctly. I've put in a fix for that, when you have a chance to try it again please let me know if that resolves the issue.

TheSast, 2023-07-02:
@nickcoutsos it seems to be working for me now! 🚀

nickcoutsos, 2023-07-02:
Great to hear!

closed 2023-07-02 by nickcoutsos