nickcoutsos/keymap-editor #120: Option to Create a Pull Request Instead of Commiting Directly to Main

jeffwilde, 2023-07-07:
I would find it very useful if the keymap-editor could could be configured to create a pull request in my repository with the keymap changes instead of directly committing to my main branch.

nickcoutsos, 2023-07-07:
Hi there!

The use case of managing keymap changes through pull requests isn't one I had really considered. I guess the main way to look at it is that repo management (creating feature branches, rolling back/rebasing commits, submitting pull requests) is a little out of the scope of the application. If I were to implement this I can imagine either doing so in a very limited fashion or else having to replicate interactions that are already available via GitHub.

In the simplest form I might have an extra action in the UI to create a new branch on your repo and switch to it so that you can easily create a PR for that once you're satisfied with your changes (which may involve multiple commits).

Could you tell me a little more about your use case? How frequently do you expect to create these pull requests?

calumy, 2023-07-07:
> Hi there!
>
>
>
> The use case of managing keymap changes through pull requests isn't one I had really considered. I guess the main way to look at it is that repo management (creating feature branches, [...]

Having the ability to create a branch without having to go to GitHub first would be great

nickcoutsos, 2023-07-08:
Thanks, @calumy. @jeffwilde my current plan is to implement that simple branching functionality because I think the cost-benefit makes it worthwhile regardless, but I'm still open to hearing more about what could work for you.

I'm not sure how quickly I'll get on that, I'm feeling like now is a good time to take a short break from the project, but when I'm ready this will be at the top of the list.

jeffwilde, 2023-07-08:
hi @nickcoutsos. That's great to hear, branches work for me as well. :+1:

For the record, the reason that I use PRs with the keymap-editor is because I have other github actions that run after the keymap-editor has made it's commit; for instance, to render an svg preview of the key layout.
If all of these commits go on the main branch, then one has to be careful about which commit is which, because the keymap and the svg preview could be out of sync if the wrong commit is chosen. If it goes into a PR (or a branch that I create a PR from), then all the action commits are collected together in the PR and squash-merged into the main branch when they're done, which ensures the main branch is always in a consistent state.

@calumy, I didn't realize that the existing implementation supports other branches. I can pre-create branches in the meantime. Thx for the tip.

nickcoutsos, 2023-07-11:
This was quickly thrown together, but I've added a "three dot" menu for extra GitHub-related actions, including branching:
[image]

[image]

You can specify a name and the source branch (defaulting to the current branch). There's no validation happening at the moment so I guess just be sensible with your branch naming conventions (I guess it'd be worth testing with slashes for branches like `feature/foo`).

nickcoutsos, 2023-07-15:
@jeffwilde I'll probably tweak the three-dot button's style at some point to match the rest of the action bar but I'm calling this done. You're welcome to re-open the issue if you have any problems.

closed 2023-07-15 by nickcoutsos