nickcoutsos/keymap-editor #121: Show the Clipboard/Download Even When Repo Connected

jeffwilde, 2023-07-09:
If the editor is in github-connected mode, the clipboard/download buttons are hidden, and the only way to access the keymap is by committing it to the target repository. I think it'd still be useful to be able to access the keymap definition within the editor UI  when connected to a repo, even if it's just to view it it.

When the repository has branch protections enabled, it's worse in that there is no way to get the keymap out of the editor at all (#120 will provide a resolution for this).

nickcoutsos, 2023-07-09:
It has crossed my mind, for sure. Actually when I first released the clipboard source this functionality existed somewhat accidentally because switching between keymap sources didn't necessarily clear out the old data.

Eventually when live keymap updates become possible I think there will still be value in being able to export the keyboard's version of the keymap to a GitHub repo or any other code-based source, if only for the benefits of revision history rather than leveraging GitHub Actions for firmware builds.

Before that, there are some fine details that would need to be sorted out to make sure that switching between sources works as expected. I think the simplest example is the fact that each source gives you access to potentially a number of keymaps for a number of keyboards, and whatever mechanism "copies" the data between the two needs to be able to verify that there isn't a mismatch. Trying to copy a keymap for a full-sized keyboard layout to a smaller ergo layout or vice versa would be a mess for sure.

Being able to at least spit out the code for the current keyboard state is a safe bet, though I might keep that a little out of the way to avoid confusion with the more typical workflows. I'll update here when I have more thoughts about how I'd like to implement it.

nickcoutsos, 2023-07-15:
Just a small update: I started to touch on an implementation here and immediately realized that part of the challenge would come from not having the original keymap source client-side (the parsed keymap data doesn't include any nodes that I don't currently manage like conditional layers or even source code comments and all of this is needed to properly merge changes back into the document).

This raises some other questions -- I can update the API endpoint to return the keymap's raw devicetree content along with the parsed data structure, but now that the app is fully capable of parsing and serializing devicetree code within the browser then should any parsing happen on the server side? I probably won't make a change as significant as that any time soon but I still want to give this some more thought before I'll get around to implementing this feature for the GitHub source.

nickcoutsos, 2023-08-26:
Alright, it's pretty basic but I've implemented a dialog that let's you view keymap changes as they would be applied to the source `.keymap` file.

[image]

[image]

closed 2023-08-26 by nickcoutsos