nickcoutsos/keymap-editor #122: Can't figure how to generate .hex or firmware files out of keymap editor

amin-mm-dev, 2023-07-12:
Just need a little help on how to generate the corresponding firmware files for ZMK using the keymap editor. I see [polarityworks](https://www.polarityworks.com/editor/index.html) uses this repo as a dev base.

Can anyone assist me with this?

Thanks!

nickcoutsos, 2023-07-12:
The app just edits keymaps, it doesn't run firmware builds. Instead it leverages GitHub actions for simplicity. So typically you would:

1. Log in to the editor with via GitHub
2. Install the app for one or more relevant repositories
    - Make sure that the repo(s) in question are ["zmk-config"-styled](https://zmk.dev/docs/user-setup#github-repo) repositories and that you've enabled actions
4. Load a keymap from a repo in the app, make changes
5. Click _Save Changes_ to commit the new keymap to your repo

This should kick off the build process. The app should detect this and link you to the new build, or you can navigate to it on GitHub via your repository's _Actions_ tab.

If you'd rather not rely on GitHub you can

- Use the _Clipboard_ source and copy/paste keymap data directly, or
- (Chromium-based browsers only) Use the _Filesystem_ source to read and write keymap data from files already present on your computer

With the local options it's up to you to handle compilation.

Maybe if you share a little about what stage of this you're starting off from I might be better able to help. You mentioned PolarityWorks, have you made a fork of one of their keyboard's ZMK config repos?

nickcoutsos, 2023-07-16:
Any luck @amin-mm-dev?

nickcoutsos, 2023-07-18:
I'm closing this for now, if you come back to this and want some help go ahead and re-open the issue or create a new one.

closed 2023-07-18 by nickcoutsos