nickcoutsos/keymap-editor #123: Github Integration Not Working

Emilostuff, 2023-07-19:
I am unable to connect to Github. When clicking "Login with Github" the site stalls for a minute and then redirects to https://zmk-keymap-editor.fly.dev/github/authorize with error HTTP ERROR 502. Everything worked fine yesterday, but today it has suddenly stopped working.

nickcoutsos, 2023-07-19:
Hi @Emilostuff, thanks for the heads-up. I haven't deployed any changes to the backend or frontend recently but it looks like my host had some issues this morning during a period that seems to coincide with your issue: https://status.flyio.net/incidents/f3jn8jh46nw6

[image]

That didn't look like fun. It looks like it's resolved, though, if you have a chance to try it again please let me know how it goes :)

Emilostuff, 2023-07-19:
That makes a lot of sense. Thanks for checking up on it! Everything works fine again :)

closed 2023-07-19 by nickcoutsos