nickcoutsos/keymap-editor #124: Trouble mapping Backspace

Peter-Col, 2023-07-23:
I'm new to your great interface, but keeps crashing when inputting backspace

```
NotFoundError

The object can not be found here.

  [native code]:undefined:undefined
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:94704
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:95949
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:97527
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96409
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96409
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96409
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96409
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96409
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96409
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:97527
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96409
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96409
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96409
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:97527
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:97527
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96409
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96409
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96409
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:97527
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96409
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96409
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96409
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:97527
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:97769
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:97555
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:97527
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:97527
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:97527
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:97527
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96347
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:97527
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96099
  https://nickcoutsos.github.io/keymap-editor/react.5a014ff6c3bc1104749a.js:2:96212
```

nickcoutsos, 2023-07-23:
Could you be more specific about what you mean by "inputting backspace"?

Peter-Col, 2023-07-24:
Sorry, I’m new to ZMK and now realised that it was not just the backspace
that was the issue. What I meant was that when using the Keymap Editor to
change the left Thumb key from the default one (which was space) on the
Corne setup, the screen would sometimes freeze with the message that I
attached. Have since encountered the same issue intermittently with other
keys eg adding the Bluetooth Select Profile 2 to another key. Not
consistently though.
Seem to have gotten through by frequent saves, and have installed the new
keymap onto my Corne.
Thanks again for creating this interface - really helpful. I hope my
message was helpful.
Kind Regards,
Peter

nickcoutsos, 2023-07-24:
I'm glad that you're having at least some success using the app, but if this is a persistent problem (I think there were ~40 such errors logged yesterday, possibly from your session) I'd really like to solve it. There have been sporadic errors before that seem like they could be related to this but you're the first person to step forward so if you're able to help me better understand what's happening it would mean a lot.

From your description it doesn't sound like the specifics of what you're doing (which key is being bound with what value) matter, and from the call stack in the error message it looks like the problem is coming up while React is doing its processing rather than in the application logic.

Can you tell me a little about your browser setup? I'm wondering if there are any extensions or even built-in functionality (to do translation, maybe) that would manipulate the DOM and leave it in an unexpected state when React attempts to re-render.

Something like this during keybind actions could mean the DOM tree changing unexpectedly at the moment that a modal dialog is being opened/closed.

nickcoutsos, 2023-07-24:
Going ahead with the idea of browser translation corrupting the DOM tree, I was able to reliably reproduce the same kind of error in my dev environment using the following steps:

1. Open Keymap Editor in Chrome (demo keyboard is fine)
2. Right-click to open the context menu and chose the translate option (I went with English to Japanese)
3. Click on the binding for `Space` to open the binding dialog
4. Click on the `Space` key code parameter to open the key code picker dialog
5. Click on the `Backspace` option (should be the second in the list)

I've pushed a fix for this instance. It's possible there are others and I'll have a look for similar areas of the codebase that could be affected.

I've also updated my error logging so that if another exception is thrown during React rendering I should also get an indication of which component to look at. Please let me know if you're encountering the failure and I can look at other potential causes.

Peter-Col, 2023-07-25:
Hi Nick,
Thanks for doing this - I’ll let you know if anything comes up; I’m an
incessant tinkerer, so will be using the editor intermittently for a while
to find that “perfect” setup!
Cheers,
Peter

nickcoutsos, 2023-07-25:
I see that a few more errors have popped up in another location, I'll be away for a few days and may not be able to address everything right away but the new instances follow the same pattern which should make it easier for me to start tracking them down proactively. I'll update here when I can!

nickcoutsos, 2023-08-04:
The error rate has been reduced since I started applying fixes to DOM manipulations such as those by translators so it looks like that strategy of testing and exception tracing is a good one.

I'm sure this isn't the end of it yet, but I'm going to go ahead and close the issue now -- if another one crops up and I'm not getting to it quickly enough feel free to make another issue to grab my attention.

closed 2023-08-04 by nickcoutsos

Peter-Col, 2023-08-04:
Hi Nick,
Thanks for letting me know. I have used it a few more times and no issues
since the last email. I’ll let you know if anything comes up. Again thanks
on behalf of all of us who benefit from the ease that your app allows!
Cheers,
Peter