nickcoutsos/keymap-editor #125: add global-quick-tap-ms to combos?

DrJA3000, 2023-07-25:
Hi, absolutely love your web-hosted zmk, it really enabled me to set-up my keyboard as a non-programmer, and am very grateful for that.

As I was adding combos, they started to cause trouble when typing quicker, and I found that zmk recently added the global-quick-tap-ms for combos as well. This should help to prevent triggering combos during normal typing, 
Admittedly, I managed to add it myself, but since I am not a programmer it took me 3 tries before I got it right. It might be a nice addition to the combo menu. 
Thx again for the work you've put in!

nickcoutsos, 2023-07-25:
Hi!

I'd love to add support but it looks like it hasn't been added yet...

- [Current binding schema for the `zmk,combos` container](https://github.com/zmkfirmware/zmk/blob/cb9c573b534dcb52ee8cf6b2cec6cb880e841301/app/dts/bindings/zmk%2Ccombos.yaml)
- zmkfirmware/zmk#1387

Looking at the schema presented there, I think I'm seeing a couple of things that may have complicated your attempts to use it

- You seem to be building from the ZMK main branch and not the PR
- The property would be defined per-combo, not in the container (I suspect when you tried this in an earlier commit the build failed because the `global-quick-tap-ms` property is not defined in the child schema for the combos container, according to the version of ZMK being used)

**How you can proceed from here**

If you update your `config/west.yml` and point to the branch from that PR (see https://zmk.dev/docs/features/beta-testing) that should allow your previous commit to build correctly when specifying the property within the individual combo's node.

With fingers crossed, you should still be able to load, edit, and save your keymap via this editor without error or loss of that unrecognized property (though it won't appear in the UI, and newly created combos will need to have it manually added to the keymap source). At least, that's how I designed behavior editing, it's entirely possible I didn't go back and fully apply these principles my much older combo editing implementation.

**How I can proceed from here**

The good news is that as soon as I hear that this feature has been merged in it should only require a data change in a couple of places on my end, and not a modification to code.

It sounds like this is highly anticipated so I sympathize with the frustration of waiting, but there are a lot of popular contributed features still going through the review process (mouse keys!) and I'm not in a good place to a) keep track of them all, and b) provide the end user with a usable way to identify which features they can use in their particular ZMK fork selection.

closed 2023-07-27 by DrJA3000

megavenik, 2023-10-03:
Hi! It's finally merged! https://github.com/zmkfirmware/zmk/pull/1387

nickcoutsos, 2023-10-03:
Perfect, I've got the necessary changes prepared in another branch, will have that deployed very soon!

megavenik, 2023-11-01:
@nickcoutsos so maybe lets reopen this issue?)

megavenik, 2023-11-01:
> @megavenik I've added the new property some time ago, have you been encountering any problems?

oh, I've been checking commits on the `main` branch and completely missed ones on the `pages`! Thanks a lot, I will give it a try:)