nickcoutsos/keymap-editor #127: Implement ZMK_MACRO convenience C macro

ctranstrum, 2023-07-28:
This is a really cool project! Looks like it doesn't currently understand the ZMK_MACRO convenience macro for declaring macros, though. https://zmk.dev/docs/behaviors/macros#convenience-c-macro

nickcoutsos, 2023-07-28:
Hi, glad you like it!

I'll try to be brief for now as I don't have a lot of time, but there are a couple of considerations here:

The first and most straightforward one is that the parser I'm using, https://github.com/joelspadin/tree-sitter-devicetree, is currently only set up to parse preprocessor function calls in certain locations. As it stands now a `ZMK_MACRO(...)` inside of the devicetree root would be treated as an error even though that is the correct placement.

The second issue is more of a preferential one. I'm not a big fan of using pre-processor macros to abstract keymaps to this extent. I should split this off into its own article but I have a writeup about some of my reasoning here: https://github.com/nickcoutsos/keymap-editor/wiki/Autoshift-using-ZMK-behaviors#why-wont-i-parse-custom-macros

`ZMK_MACRO` is probably the most justifiable use of macro functions since its built into ZMK itself, but I'll still refer to the last bit -- if you want to use these helpers to write a keymap you don't need this editor to do the same.

ctranstrum, 2023-07-28:
Oh, thanks for that excellent write up. I’ve definitely learned some things
about how to use macros. Thanks so much.

nickcoutsos, 2023-07-31:
Hi @ctranstrum, I broke that wiki article out into its [own page](https://github.com/nickcoutsos/keymap-editor/wiki/Preprocessor-Support) and added some more detail -- you don't need to go read it, but it's there. I'm going to close this issue but if you'd like to discuss it more you're welcome to re-open it or create a new one (or maybe a thread in Discussions would make more sense).

closed 2023-07-31 by nickcoutsos