nickcoutsos/keymap-editor #128: Trouble with media fast-forward/next and rewind/reverse

prudencekp, 2023-08-01:
To whom it may concern,

I've been trying to map media keys for:

1. Fast forward (next song) and I've tried fast forward (advances within the song) & next.
2. Rewind (previous song) and I've tried rewind (reverses within the song), previous, and prior.

Do you have any idea on how to map these?  Thanks!

nickcoutsos, 2023-08-01:
Hi @prudencekp 

Have a look at ZMK's documentation for [media keycodes](https://zmk.dev/docs/codes/media). Note the columns at the right that indicate OS support and that the keycodes prefixed with `C_` are generally better supported than the `K_` prefixed equivalents.

_What_ these keycodes actually do depends on how the OS chooses to interpret them, but I think what you're describing is pretty typical of `C_NEXT` (tap: next track, hold: jump forward in current track) and `C_PREV` (tap: previous track, or return to start of current track, hold: jump backward in current track).

Beyond binding these codes in your keymap (I see you're using `C_REWIND`, `K_PREVIOUS`, and `K_NEXT` currently) if you still can't get the desired effect it may be worth checking in with people in the ZMK discord server or GitHub repo.

prudencekp, 2023-08-01:
Thanks so much.  That worked great!

nickcoutsos, 2023-08-01:
Excellent! 🥳

closed 2023-08-01 by nickcoutsos