nickcoutsos/keymap-editor #129: Cannot resolve symlinks

dnlbauer, 2023-08-05:
Hi,

It would be nice it the editor would be able to follow checked-in symlinks. I.e in this repo https://github.com/dnlbauer/corax56-zmk-config/tree/main/config the keymap is linked to a the corresponding file in the boards folder and I only get "Could not parse keymap. Missing devicetree root node."

I tested if the linking is what is producing the error and it indeed is. When I remove the link and copy the file over, it loads the map successfully (as in this branch: https://github.com/dnlbauer/corax56-zmk-config/tree/no_symlink)

nickcoutsos, 2023-08-05:
Hm. I'm not sure why I thought this worked when its not something I had tested for myself. I know GitHub's API distinguishes between regular files and symlinks so as long as it's not too difficult it definitely makes sense to support. I'll review the API documentation and update here.

nickcoutsos, 2023-08-05:
https://docs.github.com/en/rest/repos/contents?apiVersion=2022-11-28#get-repository-content

The documentation seems to indicate that the endpoint should return the linked contents already but that seems to not be the case regardless of the content type I request. I'm going to have to do some digging.

dnlbauer, 2023-08-05:
Interesting. I now wonder if it has something to do with how the file is checked in. I notice that when I open the file in the github UI, it shows the path to the original file and no content. When I check it out it shows the original content.

nickcoutsos, 2023-08-05:
Oh neat. In your repo you have `config/corax56.keymap` symlinking to `config/boards/shields/ownkb/ownkb.keymap` which doesn't exist. If that's updated to the renamed keyboard's path it loads correctly!

I guess this technically follows with what the documentation described but I really expected an error response. This is especially unhelpful when requesting the raw contents, so I'd have to do an extra API call if I wanted to handle this edge case.

I'm gonna have to give that some more thought when my head is clearer than it is right now.

dnlbauer, 2023-08-05:
oh nice catch! this is really strange. I indeed renamed the files but locally, the link was still working. how strange that git didnt pick that up..

I recreated the link and its working now. Thanks for looking into this with me!

nickcoutsos, 2023-08-05:
Yeah that's exactly it. Not really a "quirk" as it seems to be by design, and I should be able to request the non-raw version and use its metadata to make sure I'm getting the data I expect.

nickcoutsos, 2023-08-11:
Pushed an update; keymap files are fetched using the JSON media type so that I can detect invalid symlinks and respond with an error about the file being missing. I should probably do this generically, and it wouldn't be much harder to handle in my abstraction, but this will do for now.

closed 2023-08-11 by nickcoutsos