nickcoutsos/keymap-editor #131: "Failed to save keymap changes" error

iamim, 2023-08-08:
Getting this error now in the online tool. It works sometimes, not sure when. I tried to log out just now and it allowed me to save once but I'm not certain if it's just a consequence.

Would love to assist it testing if needed. Thanks for the awesome tool!

nickcoutsos, 2023-08-08:
I don't see any errors logged relating to this so I would need your help to get a better idea of what's happening. It looks like you've saved changes since then, does the situation seem to have resolved itself or does it still happen intermittently? Was it specifically from clicking on the save button or were you doing something else when the error occurred? Do you have a screenshot or text copy of the error message?

nickcoutsos, 2023-08-08:
Ah, I do see some logs from a timeframe that seems to line up with your commits. I was expecting exceptions coming from the devicetree modification or the GitHub integration but it looks like instead they're `request aborted`.

There aren't any other such errors recently and I don't see anything from my host so my instinct would be a network issue.

nickcoutsos, 2023-08-10:
Gonna close this for now, haven't seen similar issues for a couple of days. If it continues to be a problem and you'd like to try some more diagnostics (although I'm not sure what besides keeping an eye on the network panel in your browser) you can re-open or create a new issue with the details.

closed 2023-08-10 by nickcoutsos

iamim, 2023-08-11:
Hi, sorry didn't reply earlier.

I tried it this morning and so far it's good. Thanks for looking into this!