nickcoutsos/keymap-editor #136: `TypeError` when using the example autoshift keymap

zakj, 2023-09-04:
Pasting the example keymap from https://github.com/nickcoutsos/keymap-editor/wiki/Autoshift-using-ZMK-behaviors renders the following error, and UI is unresponsive until I `localStorage.clear()` from the browser console.

```
TypeError

Cannot read properties of undefined (reading 'parsed')

  https://nickcoutsos.github.io/keymap-editor/LayerEditorTab.aa29e5be5e043e2a2882.js:1:20538
  https://nickcoutsos.github.io/keymap-editor/LayerEditorTab.aa29e5be5e043e2a2882.js:1:10963
  https://nickcoutsos.github.io/keymap-editor/LayerEditorTab.aa29e5be5e043e2a2882.js:1:10904
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:166:136
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:289:385
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:279:388
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:279:319
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:279:179
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:270:87
```

nickcoutsos, 2023-09-04:
Thanks for the heads up; looking now!

nickcoutsos, 2023-09-05:
Alright, looks like there's a small collection of silly mistakes and I'm not sure how this would have ever worked besides maybe some local temporary development that was deployed at some point but not actually committed.

1. I never got far in experimenting with handling parameterized macros due to the complexity in resolving their parameter types dynamically, but even worse apparently the editor wasn't recognizing nodes for parameterized macros. Just pushed a fix for this
2. The sample devicetree you linked is invalid -- I managed to miss the `>;` for the layer bindings property. Just updated it.
3. Clipboard keymap handling still needs some refinement, and as you've had the misfortune of running into. The first time you try to save keymap content it saves whatever gets parsed into localstorage and because everything else tries to default to what you'd already selected that means that reloading the page will give you the same error without a way to fix it without clearing localstorage and starting over.

zakj, 2023-09-05:
Thanks so much for this project, it's been enormously useful! Thanks for the fixes—I just worked around it with a more-trivial `&as` behavior and a bunch of `&as LS(x) x` duplication. I think agree with your general take that simpler is better here.

closed 2023-09-05 by zakj

nickcoutsos, 2023-09-05:
If you're willing to give it another shot the behaviors outlined on that page should be at least syntactically correct. Looking back I think what happened is that I worked through the theory of setting up the behaviors with another user on Discord before I had _any_ work on parameterized macros done, so I hadn't tested the final draft of code from my article.

Sorry again for the inconvenience. I'm adding a small notice in the app to point out similar layer bindings issues.

zakj, 2023-09-05:
No more `TypeError` with the updated snippet from the wiki, but it seems the macros are too complex to be well understood. Pasting the full example results in a bunch of unknown mappings:

[image]

But please don't pour any more time into this on my account! I'm quite happier with the simpler solution I'm using now. :)