nickcoutsos/keymap-editor #137: Type error while trying to override existing behavior

ammon134, 2023-09-11:
```
TypeError

Cannot read properties of undefined (reading 'value')

  https://nickcoutsos.github.io/keymap-editor/6538.60e18e71838a18bab272.js:1:19686
  https://nickcoutsos.github.io/keymap-editor/6538.60e18e71838a18bab272.js:1:21790
  https://nickcoutsos.github.io/keymap-editor/6538.60e18e71838a18bab272.js:1:21766
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:182:188
  webpack:///node_modules/react/cjs/react.production.min.js:25:207
  https://nickcoutsos.github.io/keymap-editor/6538.60e18e71838a18bab272.js:1:21744
  https://nickcoutsos.github.io/keymap-editor/6538.60e18e71838a18bab272.js:1:21673
  https://nickcoutsos.github.io/keymap-editor/6538.60e18e71838a18bab272.js:1:15138
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:166:136
```

nickcoutsos, 2023-09-11:
Hey, thanks! I think I've got it sorted out, please confirm when you get a chance.

If you're still having problems it would help to provide a little more detail about the behavior you're trying to reconfigure and maybe a sample keymap to look at.

ammon134, 2023-09-12:
Thanks for the quick fix. Seems like I'm no longer having the issue, for now. I'll keep using the tool and let you know.

And hey, thanks for making this :)

closed 2023-09-12 by nickcoutsos