nickcoutsos/keymap-editor #139: Feature request: Export layouts and settings as screenshot/PDF

t18n, 2023-09-19:
When we update the keyboard layout, having a document which can remind us of the current setup is extremely handy. I currently manually take screenshots, but I think this step can be automated. Better yet, we can add the watermark so more people can be introduced to this amazing tool.

joshsurber, 2023-09-20:
I was just coming to request this exact feature... I wish I knew enough react to try to create a PR for this

nickcoutsos, 2023-09-20:
Thanks for sharing that! I agree, it would be quite useful but not something I have on my plate right now. If you look at https://github.com/nickcoutsos/keymap-editor/issues/114 there are people using keymap-drawer which is already very featureful and if you're using keymap-editor's GitHub integration then you can have it automatically generate the image on every commit

nickcoutsos, 2023-09-27:
Hey again 👋 

Thanks again for the feedback! I've started a [discussion](https://github.com/nickcoutsos/keymap-editor/discussions/141) on this topic to get more input in one place as this is not an uncommon request.

closed 2023-09-27 by nickcoutsos

t18n, 2023-09-27:
Thanks you for your suggestion @nickcoutsos. As a temporary solution, I now use keymap-drawer to generate the keymap draw with GH workflow.
https://github.com/t18n/zmk-config-corne/blob/main/.github/workflows/draw-keymaps.yml
Would be amazing if we have all-in-one solution though. 
Thanks a lot for your amazing work on this tool. <3