nickcoutsos/keymap-editor #140: order of behavior definitions in .keymap file

z5huang, 2023-09-27:
The editor seems to refuse to save any change to a .keymap file that already has out-of-order behavior definitions (where the definition of a custom behavior A invokes another custom behavior B, but is defined earlier than B in the keymap file). ZMK itself doesn't seem to care about this ordering. Would it be possible for the editor to tolerate this as well?

nickcoutsos, 2023-09-27:
Not sure what you mean about "refuses"... would you be able to provide an example keymap that I can test? I haven't seen any errors logged about it, but the only thing I can think of where order matters is when resolving behavior parameters, and that's more of a "behavior A triggers behavior B which triggers behavior A" concern.

nickcoutsos, 2023-09-28:
@z5huang I'm seeing a number of `PayloadTooLargeError: request entity too large` messages logged, is that you? Has this been happening a lot?

z5huang, 2023-09-28:
@nickcoutsos Yes it seems to be an issue with keymap file size and unrelated to the ordering (I deleted some unused behaviors while shuffling others around trying to "fix" the problem)

nickcoutsos, 2023-09-28:
Not exactly, but there is a default limit of about 100kB. I can try bumping that up a little, but realistically there should be _some_ limit and I'd like to avoid raising it unnecessarily or so little that we keep bumping into this.

You don't need to share your keymap directly if you'd rather not, but would it be possible to get an idea of how much you're trying to include? Ie., number of layers/combos/macros/behaviors or any of that.

z5huang, 2023-09-28:
That's interesting.. the problematic keymap is only 46K (but perhaps the `#include`s are bringing in additional stuff?) That keymap has 25 layers (most of them are transparent and used as state flags), about 50 behavior + macros, and about 20 combos.  Removing one layer was enough to make it pass the size limit. Let me know if you need additional details.

nickcoutsos, 2023-09-28:
It's a little more complex than file size, as I'm parsing the `.keymap` in the backend and sending a JSON structure to the app and back. That structure only includes things that the app can edit or reference, but those things generally end up being more verbose.

Behaviors in particular can contribute a lot to that, because the structure will include both the parsed and raw versions of properties in case there's an unrecognized behavior type (like if you're building against a ZMK fork with mouse keys), so I can see 50 behaviors contributing to that.

I think what I'll do is just raise the limit to 300k or so in the morning for a little breathing room and then start doing some investigating with a larger collection of behaviors.

z5huang, 2023-09-28:
Thanks! BTW right before you ping'd me, I was using the following script to generate a dummy keymap and see if it's an issue with keymap size (after failing to create a "minimal model", due to the nature of the problem ;) ). FWIW, 100 hold-taps and 30 basic layers would already be too big
```
def gen_keymap(n_macros = 100, n_layers = 30):
  preamble = """
#include <behaviors.dtsi>
/* Include all behaviour includes needed by system behaviors/macros */
#include <dt-bindings/zmk/outputs.h>

/* Include behaviour includes needed by key binds - automatically generated */
#include <dt-bindings/zmk/keys.h>
#include <dt-bindings/zmk/bt.h>
#include <dt-bindings/zmk/rgb.h>
#include <dt-bindings/zmk/outputs.h>

/ {
    macros {
"""
  macro = """
        m%d: m%d {
            compatible = "zmk,behavior-hold-tap";
            label = "m%d";
            #binding-cells = <2>;
            flavor = "tap-preferred";
            tapping-term-ms = <200>;
            bindings = <&mo>, <&tog>;
        };
"""
            
  macros = '\n'.join( [ macro%(i,i,i) for i in range(n_macros) ] )

  keymap = """
        layer%d {
            bindings = <
 &kp F1      &kp F2    &kp F3   &kp F4    &kp F5                                                                                                &kp F6    &kp F7     &kp F8    &kp F9    &kp F10
 &kp EQUAL   &kp N1    &kp N2   &kp N3    &kp N4     &kp N5                                                                          &kp N6     &kp N7    &kp N8     &kp N9    &kp N0    &kp MINUS
 &kp TAB     &kp Q     &kp W    &kp E     &kp R      &kp T                                                                           &kp Y      &kp U     &kp I      &kp O     &kp P     &kp BSLH
 &kp ESC     &kp A     &kp S    &kp D     &kp F      &kp G                                                                           &kp H      &kp J     &kp K      &kp L     &kp SEMI  &kp SQT
 &kp GRAVE   &kp Z     &kp X    &kp C     &kp V      &kp B    &kp LSHFT   &kp LCTRL  &m0 1 1   &kp LGUI    &kp RCTRL  &kp RSHFT      &kp N      &kp M     &kp COMMA  &kp DOT   &kp FSLH  &kp PG_UP
 &m0 1 1     &kp HOME  &kp END  &kp LEFT  &kp RIGHT           &kp BSPC    &kp DEL    &kp LALT  &kp RALT    &kp RET    &kp SPACE                 &kp UP    &kp DOWN   &kp LBKT  &kp RBKT  &kp PG_DN
            >;
        };
"""
  keymaps = '\n'.join( [keymap%i for i in range(n_layers) ] )

  res = preamble + macros + '\n    };' + """

    keymap {
        compatible = "zmk,keymap";
""" + keymaps + """
    };
};
"""

  return res

import sys
if __name__ == '__main__':
    print(gen_keymap(*[ int(a) for a in sys.argv[1:3] ]))
```

nickcoutsos, 2023-09-28:
Nice, thanks for that!

nickcoutsos, 2023-09-28:
Bumped the limit a little bit. I'll keep posting here with updates if I find anything interesting or make any optimizations. Couple of notes before I forget:

First, if you _do_ run into that limit again (or some other error) I'd hate for you to lose a lot of effort -- if you haven't seen it already, there's an "escape hatch" in the form of a dropdown menu option to export your current keymap state:

[image] [image]

It applies your local changes to the version of the keymap that was last loaded and spits out the result without attempting to send it to the server. Assuming the issue in that situation is the size of the payload and not a bug in the serializer you would still have that issue when attempting to save again, but at least you won't have to start from scratch.

Second, that's quite a few layers. I remember someone talking about using lots of conditional layers as a form of state management but I'm not sure how common that is in practice... all that to say it's unexpected. How helpful have you found the editor to be with that many layers? I know that depending on your viewport size you can run into issues with the layer switcher becoming inaccessible, I'll have to do some thinking about that. Is there anything else limiting you there?

z5huang, 2023-09-29:
Thanks, the clipboard function is most welcome!

The vertical layer list does get a bit too long, but since zmk only supports 32 max, it's not too terrible (I'm not sure if making it into 2 columns would help since layer names collapse dynamically). Incidentally on my screen, the link to the github repo is blocking access to one layer.

The editor is still very useful even with many layers because it enables you to make minor changes quickly. The ability to shuffle layers around and having the editor taking care of layer references is extremely useful . (I've occasionally run into situations where the new layer number is not quite right but haven't got around to produce a minimal example. May have something to do with conditional layers)

One thing that might be useful is if after going into behaviors etc and then back to the layer view, we can return to the layer that was being worked on. Right now we are dropped into default layer. It's very minor but with many layers, I sometimes forget which one I was working on ..

nickcoutsos, 2023-09-29:
This is super useful feedback!

The GitHub link blocking the layers is what I was referring to, I'll have to think about a proper fix for that. I had thought about making the list scrollable but multiple scrollbars can be unpleasant and could make the drag-and-drop even more awkward.

Good point about remembering the selected layer. Right now when you switch tabs React is completely unmounting the layer view, and the active layer state is local to that. This has been useful for things like switching keymaps, but I definitely get how losing your position in a single larger keymap like that can break your flow.

nickcoutsos, 2023-09-30:
@z5huang I've made some updates to the layer switcher, thanks again for bringing those issues to my attention!

I'm working on something now to hopefully cut down on unnecessarily large request payloads. When keymap changes are being sent back to the server the original keymap data is still available client-side, so I can match up any nodes with their original version and compare the two. If they're equal I can mark it as such and omit the node from the payload.

It's about as coarse as a diff can get: if you change a single binding or even the label in a layer then that entire layer node gets included. At some point I may _need_ something more granular, possibly when ZMK has a protocol for runtime keymap updates and I can only change one item (like... one binding cell? who knows) at a time, but for now I want see if I can get some decent savings with minimal changes to avoid accidentally mucking things up.

It would be helpful to get an idea of how many such things you generally change at one time. I still need to do some experiments to get a feel for how much is "too much" but it's always great to have some more insight into people's workflows :)

z5huang, 2023-09-30:
Thanks for the update @nickcoutsos , keep up the great work!

In terms of typical usage, most often I will just be tweaking very minor things, like a few keys in a single layer, sometimes in conjunction with new or modified custom behavior / macro / combos. Adding entire new layers is much less frequent. 

Behavior / macro / combo are definitely the most useful features for myself as they eliminate much of the need to consult zmk docs

nickcoutsos, 2023-09-30:
Basic testing (locally) seems promising! A keymap with 240 behaviors, 30 layers, and a handful of combos and macros takes 180kb to send in the full structure, but making changes to a few layers and behaviors gives me an "abridged" update payload of just 16kb :)

The API is updated to handle the smaller payloads, I'll be making the appropriate changes in the app and deploying shortly!

nickcoutsos, 2023-10-02:
I'm not seeing any errors from regular use crop up from this change, which is always fantastic. I'm not sure if you've had a chance to work with a larger keymap since that deploy but I'm inclined to call this done based on the initial tests?

I'll leave this open a bit longer unless you want to confirm it yourself.

z5huang, 2023-10-02:
Did some test on the "problematic" keymap, no issue so far

closed 2023-10-02 by z5huang

nickcoutsos, 2023-10-02:
Thanks again for sharing this and helping me break things down! If you come across that layer-reordering issue again please do let me know :)

nickcoutsos, 2023-10-06:
This is interesting: it seems like there was an existing intermittent issue that was exacerbated somewhat by this fix. From what I can tell, GitHub isn't _super_ great about giving me the correct cache headers for a resource shortly after it's been updated, so it's possible to get into this sequence of events:

1. Load keymap latest ("version 1")
2. Add new behavior (or whatever)
3. Apply changes to latest ("version 1"), commit ("version 2"), return updated keymap state
4. Make unrelated changes
5. Apply changes to latest ("version 1" 🚨), ...

What's happening at the beginning of the "apply changes" step is my backend is requesting the current version of the keymap _unless_ it's unchanged from the cached version. Sometimes GitHub incorrectly reports that I can continue with my cached version, so a payload from the app which says the new behavior is unchanged doesn't actually have those details anymore and can't apply them to the original keymap.

I've updated this caching to force invalidation of the relevant resources when I'm changing them, and fortunately this is easy enough to do right now, but this only works because I don't need to scale to multiple servers. That said, I don't think it's worth removing the cache altogether, their API responses are slow enough already.

z5huang, 2023-10-08:
Is github aware of the discrepancy between the reported vs the actual headers? That sounds like a consequential bug on their end

nickcoutsos, 2023-10-09:
I'm not sure, really, but maybe? I think you could argue it from the perspective of it being "eventually consistent" and... I do understand _why_ it may not work perfectly, the nature of caching being what it is.