nickcoutsos/keymap-editor #142: Sofle default layers missing keys

B1gFuz, 2023-09-30:
When setting up my sofle the right board is missing 2 keys on the default layers and right encoder on right board on all layers other then layer0  when adding layers the two missing keys are there.
[image]
[image]

nickcoutsos, 2023-10-01:
Hi, thanks for reaching out!

Looking at your repo's history it appears that you've copied a keymap for the lily58 keyboard and treating it as sofle. In general that's not a big deal, but the sofle is expecting two more keybinds than are listed in your keymap's `bindings` array for that layer.

The app isn't expecting this and calling it out or giving you a way to deal with it because it's kind of assumed that you have a working base keymap already -- there isn't a use case for "adding" or "removing" keybinds from a layer arbitrarily. The other part of it is that _when_ this is encountered by the app it's frequently a result of people using aliases to replace multiple keybinds with a single symbol and that throws everything off.

**Simplest solution:** either add a couple of bindings (doesn't matter what, `&none &none` is fine) to your base layer manually and then reload in the app to change them as desired, or start from the [ZMK default sofle keymap](https://github.com/zmkfirmware/zmk/blob/main/app/boards/shields/sofle/sofle.keymap) instead. Without looking too closely at your keymap, I think you can probably get away with just replacing the last line of that bindings with the one from ZMK's sofle:

```dts
                 &kp LGUI &kp LALT &kp LCTRL &mo 1  &kp RET    &kp SPACE &mo 2  &kp RCTRL &kp RALT  &kp RGUI
```

> and right encoder on right board on all layers other then layer0

Sensor bindings are kind of the exception to what I said above. Missing bindings from the `sensor-bindings` array (or an omitted property altogether) is treated like `&trans` so the sensor bindings from the next active layer are used.

IIRC, `&trans` _can't_ be used in `sensor-bindings`, which is why they're left out when creating a new layer. What you're seeing in that second screenshot is the list of available sensors and a button to add a binding for that sensor to the list. Since you can't "skip" one, adding a binding for `RIGHT_ENCODER` on that layer means you _must_ also add one for `LEFT_ENCODER`.

B1gFuz, 2023-10-01:
thank you I was thinking that was the issue.  I am new at this and i searched all over for the keymap i will add that and restart.  Thank you for your time.

nickcoutsos, 2023-10-01:
Cool, happy to help! Looks like you've made some updates via the keymap editor now which is great, please let me know if you run into any more issues!

closed 2023-10-01 by nickcoutsos

B1gFuz, 2023-10-01:
I have been trying to figure it out on my own but it is not creating my firmware for the board and I can not figure out how to get it to work.

nickcoutsos, 2023-10-01:
Ah! Keymap editor relies on repos being set up to build with GitHub actions. You're most of the way there, but you need to enable actions in your repository settings.

You should be able to do so at https://github.com/B1gFuz/zmk-comfig-sofle/settings/actions

From there, select **Allow all actions and reusable workflows** and click save.

B1gFuz, 2023-10-01:
thank you so much.  I am trying to learn this stuff.  thank you for all the help.

nickcoutsos, 2023-10-01:
No sweat, it definitely takes some time to make sense of all the pieces and start to understand how they fit together!

nickcoutsos, 2023-10-01:
@B1gFuz I just took a peek at your build and keymap, you've got a macro where one of the `&kp` binds is missing its parameter. I _believe_ that's what's tripping things up.