nickcoutsos/keymap-editor #143: Add globe keycode

AliQ80, 2023-10-02:
hello,

ZMK has introduced a globe keycode 3 months ago, I would like to suggest adding it to the keymap editor as it seems to not be available there (or I have missed it).

[zmk documentation](https://zmk.dev/docs/codes/applications) shows it in the last row in the first table (application control) as:
GLOBE, C_AC_NEXT_KEYBOARD_LAYOUT_SELECT : AC Next Keyboard Layout Select (Apple Globe) Consumer AC

thank you

nickcoutsos, 2023-10-02:
If you mean this:

[image]

it was only three days ago :)

nickcoutsos, 2023-10-02:
Unrelated, I see you've been using the app with a reviung for some time now! If you're at all interested, I've more recently updated the layout data to more accurately represent the layout key rotation and sizing:

[image]

If you'd like to use it you should be able to delete `config/reviung41.json` and the new layout would be picked up the next time you use the app.

closed 2023-10-02 by nickcoutsos

AliQ80, 2023-10-03:
Hi @nickcoutsos , I apologize for taking your time, you are correct it was 3 days ago, I don't know how I misread that,

I also don't know why globe didn't show up when I tried before.

Thank you for pointing out the reviung41 lay out, I will check it very soon

I really appreciate your work on this project, it's very under rated, and should have way much more recognition.