nickcoutsos/keymap-editor #144: Feature Request : Add duplicate button

EpiCanard, 2023-10-03:
Hello

I suggest the following feature : 
For the views: macros, behaviors and conditional layers, add a button duplicate.
When we click on it, a popup window opens, just like when we create a new element, but with fields already filled in with the information of the element we want to duplicate.

We can save a lot of time and avoid repetitive manual actions on the part of the user.

Best Regards

nickcoutsos, 2023-10-03:
Hi, thanks for sharing!

It's an interesting idea, but one thing that makes me pause is the fact that all of these features have at least one element that must be unique between them (the node name, and in the case of behavior types, the devicetree reference and label properties) and other elements that _should_ be unique to make it useful.

Do you have some example use-cases of how you'd duplicate a node and make it sufficiently unique to be useful? On thing I'd imagine is perhaps conditional layers where a number of the `if-layers` values are common but it's easier to just delete some irrelevant layers from the duplicated node instead of adding everything.

EpiCanard, 2023-10-03:
The button will not really duplicate the node but a open the creation node popup with elements already pre-filled so nothing will be created until the user has clicked on "create".

_For example,_
You want to create a behavior "mod-morph" to override the behavior of left shift when you press a key.
If you want to create another behavior with the same mods and with bindings slightly different, actually you will have to fill all the information again.
The idea is to be able to click on a icon duplicate and it will open the Edit behavior popup, it will consider that  mod-morph has already been selected, and it will fill all fields with values of the behavior we want to duplicate.
With same bindings, same mods and same keep-mods.

EpiCanard, 2023-10-03:
Each view seems to have its own workflow, so I guess it will not be an easy feature.
But it could be really a time saving feature when you create a lot of similar macro and behavior.

nickcoutsos, 2023-10-03:
Yeah, not unreasonable. I think I can start with the behavior version, see how well it works out, and decide if/how best to implement it across other features.

Thanks!

nickcoutsos, 2023-10-04:
Alright this worked out really nicely: because of the initial behavior type selection I already have an easy way to handle these transient behavior records in the editing dialog, so adding the clone button only required a handful of newlines in one file.

[image]

I need to go back and add validation to the input fields for node name/label/bind to stop people from sticking with the default `new_behavior` values but that concern is separate from this one already.

nickcoutsos, 2023-10-04:
... and now I've added it to combos, macros, and conditional layers.

I'm going to consider this done for now, but there are some improvements that could be made. The patterns for browsing and editing have evolved throughout development as new features and requirements were added. At some point I'll be making some refactors to get everything using the same components for managing things like node names and properties and will likely take that opportunity to get some consistency in the new/clone dialog and workflow as well.

Let me know how you like the new feature!

EpiCanard, 2023-10-05:
That was fast !

This feature is really useful and works really well !
It doesn't prevent from using the features, but it's true that new/clone dialogs lack a little consistency between workflows.

Thank you very much for your work !

For me I consider this feature request as done, if it's ok for you, you can close it.