nickcoutsos/keymap-editor #147: Feature request: parsing #ifdef statements in .keymap files

jlnewman01, 2023-11-02:
A couple board definitions such as the out-of-tree technikable: https://github.com/petejohanson/technikable-zmk-config/blob/main/boards/arm/technikable/technikable.keymap and the in tree BT60/CKP boards https://github.com/zmkfirmware/zmk/blob/main/app/boards/arm/ckp/bt60_v2.keymap use #ifdef statements in the keymap file to support multiple layouts, as it stands the editor doesn't know what to do with them and so gets confused at the presence of lots of identically named layers. Would it be doable to add support for these statements?
An example of such a keymap file in a config repo: https://github.com/jlnewman01/zmk-config/blob/master/config/bt75_v1.keymap
The file can still be edited and works fine, but the presence of 8 layers and all the warnings was a little confusing
Thanks :)

nickcoutsos, 2023-11-02:
I don't think it's likely that it'll be added. (Please see [Preprocessor Support](https://github.com/nickcoutsos/keymap-editor/wiki/Preprocessor-Support#what-are-the-challenges-to-supporting-preprocessor-macros) in the wiki for some of the details about the complexity).

If it's not possible for you to run it through a preprocessor locally to evaluate those macros, you can edit the keymap file by hand and remove the alternative branches.

nickcoutsos, 2023-11-02:
For the BT75 example, another approach would be board-specific keymaps (e.g. `bt75_iso.keymap`) that you edit directly and keep the preprocessor branching in `bt75.keymap`. But that still boils down to doing the work ahead of time (ideally by the vendor).

closed 2023-11-02 by nickcoutsos