nickcoutsos/keymap-editor #149: Miss-mapped keys during quick `&kp` assignment

Samwell-II, 2023-11-03:
While using the quick assign `&kp` feature the left brace is switched with the right bracket when pressed.
Ie, I want to assign a key to `{` so I press `shift + {` and the key is assigned to `shift + ]`. Similarly, I wish to assign a key to `]`, so I press `]`, and the key is assigned `{`.

As an asside, Thank you for providing this tool!

nickcoutsos, 2023-11-03:
That is weird and interesting. I'll have to dig deeper to see where precisely my mapping is going wrong but I can confirm the issue. Thanks for the heads up, I'll get you an update as soon as I know more!

nickcoutsos, 2023-11-04:
Turns out it was just a dumb copy and paste issue. Should be fixed now!

closed 2023-11-04 by nickcoutsos