nickcoutsos/keymap-editor #151: Feature Request: Add ErgoDox layout

justbest23, 2023-11-07:
Heyo! Any chance you could add the Ergodox layout? Here's an image of the layout:

[image]

Thanks!

nickcoutsos, 2023-11-07:
Hi @justbest23, I actually took a look at this a while back -- that's SliceMK's Ergodox with the dongle, right?

It felt weird to me that the dongle's button is included in the key scan matrix and wasn't sure if it would lead to problems down the road. I believe that "key" is hidden in their configurator but because it's part of the matrix transform it has to be part of the bindings array and I can't ignore it without making a special case for this keyboard.

Here's a JSON layout you can use to achieve this layout:

```json
{
  "id": "slicemk_ergodox",
  "name": "SliceMK Ergodox",
  "layouts": {
    "default_layout": {
      "layout": [
        { "label":      "dongle", "row": 0, "col":  0, "x": 9.25, "y":    0, "w": 1.5, "h": 0.5 },
      
        { "label":         "Esc", "row": 1, "col":  0, "x":    0, "y": 0.33, "w": 1.5 },
        { "label":           "1", "row": 1, "col":  1, "x":  1.5, "y": 0.33 },
        { "label":           "2", "row": 1, "col":  2, "x":  2.5, "y": 0.12 },
        { "label":           "3", "row": 1, "col":  3, "x":  3.5, "y":    0 },
        { "label":           "4", "row": 1, "col":  4, "x":  4.5, "y": 0.12 },
        { "label":           "5", "row": 1, "col":  5, "x":  5.5, "y":  0.2 },
        { "label":           " ", "row": 1, "col":  6, "x":  6.5, "y":  0.2 },
        { "label":           " ", "row": 1, "col":  8, "x": 12.5, "y":  0.2 },
        { "label":           "6", "row": 1, "col":  9, "x": 13.5, "y":  0.2 },
        { "label":           "7", "row": 1, "col": 10, "x": 14.5, "y": 0.12 },
        { "label":           "8", "row": 1, "col": 11, "x": 15.5, "y":    0 },
        { "label":           "9", "row": 1, "col": 12, "x": 16.5, "y": 0.12 },
        { "label":           "0", "row": 1, "col": 13, "x": 17.5, "y": 0.33 },
        { "label":           "-", "row": 1, "col": 14, "x": 18.5, "y": 0.33, "w": 1.5 },
      
        { "label":         "Tab", "row": 2, "col":  0, "x":    0, "y": 1.33, "w": 1.5 },
        { "label":           "Q", "row": 2, "col":  1, "x":  1.5, "y": 1.33 },
        { "label":           "W", "row": 2, "col":  2, "x":  2.5, "y": 1.12 },
        { "label":           "E", "row": 2, "col":  3, "x":  3.5, "y":    1 },
        { "label":           "R", "row": 2, "col":  4, "x":  4.5, "y": 1.12 },
        { "label":           "T", "row": 2, "col":  5, "x":  5.5, "y":  1.2 },
        { "label":           " ", "row": 2, "col":  6, "x":  6.5, "y":  1.2,           "h": 1.5 },
        { "label":           " ", "row": 2, "col":  8, "x": 12.5, "y":  1.2,           "h": 1.5 },
        { "label":           "Y", "row": 2, "col":  9, "x": 13.5, "y":  1.2 },
        { "label":           "U", "row": 2, "col": 10, "x": 14.5, "y": 1.12 },
        { "label":           "I", "row": 2, "col": 11, "x": 15.5, "y":    1 },
        { "label":           "O", "row": 2, "col": 12, "x": 16.5, "y": 1.12 },
        { "label":           "P", "row": 2, "col": 13, "x": 17.5, "y": 1.33 },
        { "label":          "\\", "row": 2, "col": 14, "x": 18.5, "y": 1.33, "w": 1.5 },
      
        { "label":        "Caps", "row": 3, "col":  0, "x":    0, "y": 2.33, "w": 1.5 },
        { "label":           "A", "row": 3, "col":  1, "x":  1.5, "y": 2.33 },
        { "label":           "S", "row": 3, "col":  2, "x":  2.5, "y": 2.12 },
        { "label":           "D", "row": 3, "col":  3, "x":  3.5, "y":    2 },
        { "label":           "F", "row": 3, "col":  4, "x":  4.5, "y": 2.12 },
        { "label":           "G", "row": 3, "col":  5, "x":  5.5, "y":  2.2 },
        { "label":           "H", "row": 3, "col":  9, "x": 13.5, "y":  2.2 },
        { "label":           "J", "row": 3, "col": 10, "x": 14.5, "y": 2.12 },
        { "label":           "K", "row": 3, "col": 11, "x": 15.5, "y":    2 },
        { "label":           "L", "row": 3, "col": 12, "x": 16.5, "y": 2.12 },
        { "label":           ";", "row": 3, "col": 13, "x": 17.5, "y": 2.33 },
        { "label":           "'", "row": 3, "col": 14, "x": 18.5, "y": 2.33, "w": 1.5 },
      
        { "label":      "LShift", "row": 4, "col":  0, "x":    0, "y": 3.33, "w": 1.5 },
        { "label":           "Z", "row": 4, "col":  1, "x":  1.5, "y": 3.33 },
        { "label":           "X", "row": 4, "col":  2, "x":  2.5, "y": 3.12 },
        { "label":           "C", "row": 4, "col":  3, "x":  3.5, "y":    3 },
        { "label":           "V", "row": 4, "col":  4, "x":  4.5, "y": 3.12 },
        { "label":           "B", "row": 4, "col":  5, "x":  5.5, "y":  3.2 },
        { "label":           " ", "row": 4, "col":  6, "x":  6.5, "y":  2.7,           "h": 1.5 },
        { "label":           " ", "row": 4, "col":  8, "x": 12.5, "y":  2.7,           "h": 1.5 },
        { "label":           "N", "row": 4, "col":  9, "x": 13.5, "y":  3.2 },
        { "label":           "M", "row": 4, "col": 10, "x": 14.5, "y": 3.12 },
        { "label":           ",", "row": 4, "col": 11, "x": 15.5, "y":    3 },
        { "label":           ".", "row": 4, "col": 12, "x": 16.5, "y": 3.12 },
[... 10 lines cut ...]
        { "label":       "LCtrl", "row": 5, "col": 12, "x": 16.5, "y": 4.12 },
        { "label":       "LCtrl", "row": 5, "col": 13, "x": 17.5, "y": 4.33 },
        { "label":       "LCtrl", "row": 5, "col": 14, "x": 18.5, "y": 4.33 },
      
        { "label":       "LCtrl", "row": 6, "col":  5, "x":  7.5, "y":  2.7,                     "r":  25, "rx":  5.5, "ry": 3.7 },
[... cut ...]
        { "label":        "LAlt", "row": 7, "col": 10, "x": 12.5, "y":  3.7,           "h":   2, "r": -25, "rx": 14.5, "ry": 3.7 },
      
        { "label":           " ", "row": 8, "col":  6, "x":  8.5, "y":  4.7,                     "r":  25, "rx":  5.5, "ry": 3.7 },
        { "label":           " ", "row": 8, "col":  8, "x": 10.5, "y":  4.7,                     "r": -25, "rx": 14.5, "ry": 3.7 }
      ]
    }
  }
}
```

If it works for you, let me know and I can make it available in the app automatically.

justbest23, 2023-11-08:
Heyo! Thanks for the quick response.
 
The json you provided seems to work perfectly and I'm in the process of doing additional testing.

Is the dongle's key that bootloader button? I haven't seen that included on the other configurator I was using before.

Will integrate with github and try flashing a few configs and let you know!

From my initial import of my current keymap there were a few keys that didn't get picked up correctly. I'm not sure if that's something you can or want to fix though. It doesn't bother me too much. I'll provide the list in a follow up comment once I've done some more testing.

justbest23, 2023-11-08:
| Imported Key  |  Expected Key | Corrected Key |
| ------------- | ------------- | ------------- |
| GRAV		| `  |	GRAVE
| Content Cell  | Content Cell  |

nickcoutsos, 2023-11-08:
> Is the dongle's key that bootloader button? I haven't seen that included on the other configurator I was using before.

Right, the other configurator is tailor made for that keyboard so it has special handling for that "key" but if you look at the keymap it's part of the bindings.

> From my initial import of my current keymap there were a few keys that didn't get picked up correctly. I'm not sure if that's something you can or want to fix though. It doesn't bother me too much. I'll provide the list in a follow up comment once I've done some more testing.

I did have a look. I think in one case there was an incorrect keycode (or at least not standard? I didn't check carefully), but mostly it's the `ZMK_MACRO(...)` which I don't support (See [Preprocessor Support](https://github.com/nickcoutsos/keymap-editor/wiki/Preprocessor-Support)). It shouldn't be difficult to re-implement them with regular devicetree syntax (or graphically in the editor, just don't forget to use unique names or delete the originals). You could also have a single parameterized macro that you can reuse for every layer+colour.

justbest23, 2023-11-08:
Yea I'm not worried about the macros. In my case they're actually just "Switch to layer and also change colour so I know what layer I'm on". There is not yet support for swapping colours and layers at the same time without macros.  

Here's the few keys that didn't import correctly:

| Imported Key  |  Expected Key | Corrected Key |
| ------------- | ------------- | ------------- |
| GRAV		| `  |	GRAVE |
| PRSC	 | PrintScreen  | PRINTSCREEN | 
| BKSP | Backspace | BACKSPACE
| M_VOLD | Volume Down | C_VOL_DN |
| M_VOLU  | Volume Up | C_VOL_UP |
| M_MUTE | Mute | C_MUTE |
| PGUP | Page Up | PAGE_UP |
| PGDN | Page Down | PAGE_DOWN |
| SPC | Space | SPACE |

Also I'm sure I'm doing something wrong here, but I've linked to my repo with my keymap in it and I've put that JSON that you sent me in the same folder as the keymap (config/ergodox.json) but the editor is still saying no layout available.

Also also apologies for all the questions. I'm very new to the whole custom keyboard things so I'm still figuring out the community and how to ask questions and stuff.

justbest23, 2023-11-08:
Copying and pasting the keymap seems to work perfectly. So the layout is correct.

nickcoutsos, 2023-11-08:
> Here's the few keys that didn't import correctly:
> [...]

Are the keys in the left column from QMK? The right column seems to be the correct ZMK equivalents.

Regarding the JSON file, it should either be `info.json` (for legacy reasons) or ideally `<keyboard name>.json`. It needs to match the keymap file you're editing so in your case that would be `slicemk_ergodox.json` rather than `ergodox.json`.

Anyway, I'm happy to help out so if you're running into trouble with anything please ask away!

justbest23, 2023-11-09:
The keys in the left column come from my keymap that was generated here: https://config.slicemk.com/ergodox/

Now that I've changed the json to info.json it works fine! Thanks for the help!

nickcoutsos, 2023-11-11:
Hey, I'm sorry I missed this comment. Based on commit dates it looks like you have things in order, the simple `&mt LC(C) C` should do the trick.

A custom hold-tap like your `c_copy` should generally be OK but I think your issue is with the bindings you've selected. `bindings = <&mt>, <&kp>` means the hold part of the behavior is itself triggering a hold tap which isn't what you're after, I think, but can cause unintended problems. You can use `bindings = <&kp>, <&kp>` and it'll still accept the `LC(C)` hold parameter.

closed 2023-11-16 by nickcoutsos