nickcoutsos/keymap-editor #155: Bug: Focus is lost when using search

hallundbaek, 2023-11-19:
What I did:

I pressed on a key, and then the keycode, this opened a search dialogue and I typed in "space". I then press Enter.

What I expect:

The best match of the search term should be focused, and when I press Enter, the focused item should be picked.

What happened:

The results show the space key, but it doesn't have focus, so nothing happens when I press enter. If there are more results, one of them will have focus, and when there are a lot of results it seems to just pick one in the middle.

Browser: Firefox 95.0.1 (64-bit)

nickcoutsos, 2023-11-19:
Yeah, this is a problem, thanks for the report!

I think it might be maintaining the highlight on the position that was _previously highlighted_, even if it's not the current "best" match or the results are filtered down to a set that doesn't include that position. The goal was to show the current selection when that panel opens up but it would probably be worth clearing that once you start typing.

hallundbaek, 2023-11-20:
> The goal was to show the current selection when that panel opens up but it would probably be worth clearing that once you start typing.

That makes sense. But yeah, once you start typing you are implicitly going for a change, and if you need to press enter anyway to have it select it, it can just revert to the current selection when the search field is empty.

When searching, focus should in my opinion always be on the top element, and the search results should be ordered by closeness of match, which I think they are? I can see you can change focus by using arrow up/down, which is neat! But it highlights both the current bound option and the current selected option with the same color, which is a bit confusing.

nickcoutsos, 2023-11-22:
This should be fixed now, let me know if there's still any lingering problem.

hallundbaek, 2023-11-24:
It is certainly fixed! Great stuff!

But it highlighted another small ux issue, in that when you select behaviour for a key and press enter, the dialogue box closes, and in most cases you want to set the key straight after, so it would be nice to be able to press tab, to reach the next empty binding. But thats definitely not related to this issue.

closed 2023-11-24 by nickcoutsos