nickcoutsos/keymap-editor #156: Feature/UX: Improved naming capabilities

hallundbaek, 2023-11-19:
How you chose names for all the different elements vary, it would be nice if it was all done in the same manner.

In my opinion, whenever there is a name:

1. Layer name in top left of a layer page.
2. Cell in condition column of conditional layers tab
3. Top left in each of the combos in the combo tab
4. Binding names shown in macro tab
5. Cell in binding column of behaviors tab

It should simply be an editable input field, no need to press save or enter, it should just persist whatever is written, unless there's an overlap, in which case it should not persist it and show a small error.

Ideally it should then also rename all calls to that, so I don't have to go through all the places that used the old name and rebind them.

nickcoutsos, 2023-11-19:
> Ideally it should then also rename all calls to that, so I don't have to go through all the places that used the old name and rebind them.

This is the case for behaviors, but it looks like it's not getting applied to macros right now so I'll have to review that. I don't think anything else is reference-based so there wouldn't be anything to update besides the name.

> It should simply be an editable input field, no need to press save or enter, it should just persist whatever is written, unless there's an overlap, in which case it should not persist it and show a small error.

I did have a component for an inline editable field, I may look into reintroducing it for some of the list views. If I did, though, I would probably still need you to confirm with enter or clicking a button, especially for bind labels where changing the value has keymap-wide side effects.

hallundbaek, 2023-11-20:
> This is the case for behaviors, but it looks like it's not getting applied to macros right now so I'll have to review that. I don't think anything else is reference-based so there wouldn't be anything to update besides the name.

Ah cool! Should have checked that, my apologies!

> If I did, though, I would probably still need you to confirm with enter or clicking a button, especially for bind labels where changing the value has keymap-wide side effects.

Why? if it automatically changed references across the keymap, there shouldn't be a need for the user to confirm. And even if the references are not changed, I don't see how forcing the user to press enter really changes much?

nickcoutsos, 2023-11-21:
Macro references should now auto-update like the rest of behaviors.

Regarding pressing enter or otherwise confirming a change, it just feels like something that can have significant side effects like that should have an interaction to complete the process. Without it I'm either scanning every possible node for references or including a debounce/throttle for efficiency. There are other options like not committing the change until the input loses focus, but if that happens while a user changes from the macros tab to behaviors they may not catch an indication that it didn't complete.

Maybe I'm underestimating how often someone is renaming an element, but if your hands are on the keyboard anyway hitting enter is one more keystroke, right?

hallundbaek, 2023-11-24:
> Regarding pressing enter or otherwise confirming a change, it just feels like something that can have significant side effects like that should have an interaction to complete the process.

Maybe its just my sensibilities, but the big 'save' button is the only committing button I need. I expect everything to be saved to memory when editing, and nothing to be committed until 'save' is hit.

> Without it I'm either scanning every possible node for references or including a debounce/throttle for efficiency. There are other options like not committing the change until the input loses focus, [...]

My response here would be that it should just also save when switching tabs, i.e. any time focus is lost.