nickcoutsos/keymap-editor #157: Feature request: Drag to move elements around

hallundbaek, 2023-11-19:
It would be great if it was possible to drag a key on top of another to have them swap, this would have saved me a lot of time in the beginning when I was trying out many different things.

It would also be amazing if you could reorder conditional layers/combos/macros/behaviors in their respective lists, right now the only way to achieve it through the UI is by creating a new/deleting, which is not a viable option. I end up having to edit the code.

nickcoutsos, 2023-11-19:
I hadn't considered re-ordering conditional layers, do they have a precedence for evaluation? Out of curiosity, how complex is your setup?

Drag-and-drop is _pretty_ annoying to add and if I were to include it in more collections it would probably take the form of a "move this above ..." type dropdown or a more dynamic sort without actually relocation things at the data layer.

Dragging bindings around in a layer is interesting. Still a pain to implement but I can see some use. Could you tell me more about your workflow? Just want to make sure I'm not missing anything about the use case that could benefit from an alternate solution.

hallundbaek, 2023-11-20:
> I hadn't considered re-ordering conditional layers, do they have a precedence for evaluation? Out of curiosity, how complex is your setup?

I don't think they do? For me its mostly just to group things together for easier overview.

> Drag-and-drop is pretty annoying to add and if I were to include it in more collections it would probably take the form of a "move this above ..." type dropdown or a more dynamic sort without actually relocation things at the data layer.

A "move this above ..." type dropdown would be plenty sufficient!

> Dragging bindings around in a layer is interesting. Still a pain to implement but I can see some use. Could you tell me more about your workflow? Just want to make sure I'm not missing anything about the use case that could benefit from an alternate solution.

I have some combos that do a momentary layer, that layer then has keys bound for the "free fingers", if I change the combo keys, such that the "free fingers" change position, the layer has to get those keys moved to new positions so I can reach them when pressing the combo.

Here it would be very nice to simply drag the keys around, to make it fit the new combo. Instead of having to set them to transparent and set new keybindings, remembering exactly what is deleted, which is especially difficult if swapping two bound keys, since you'll lose the content of one once you've rebound it.

I think the real use-case if for first time users trying ZMK, because they're probably going to want to try a lot of different things, but usually with the same bindings, just the positions changing. For example, I have a columnar layout and short pinkies, so I've shifted the "qaz" and "0p;" columns down and back a few times to find the right feel, and that operation was quite cumbersome and would have been greatly helped by being able to swap keys.

nickcoutsos, 2023-11-21:
Yeah, I can see the benefit there. By "free fingers" I suppose you mean something like activating a layer on one half of the keyboard to change keys on the opposite half, and then flipping things around?

hallundbaek, 2023-11-24:
> Yeah, I can see the benefit there. By "free fingers" I suppose you mean something like activating a layer on one half of the keyboard to change keys on the opposite half, and then flipping things around?

Hmm not exactly, I have a combo on 'k' + 'l' which enables a layer that allows me to do parenthesis on 'j' and 'm', if I were to move that combo up to 'i' + 'o' then I'd want to move keys in that layer to 'u' and 'j', which takes one setting to &trans, and one setting &kp and key, and one key rebind. Ideally I'd just drag the keys around instead.