nickcoutsos/keymap-editor #158: Bug: using `&macro_param_1to1` for the second time in a parameterized macro fails to include the second `MACRO_PLACEHOLDER`

z5huang, 2023-11-22:
How to reproduce:

1. Add a new macro. It will start with a `&none`
2. Add a `&macro_param_1to1`
3. Move the `&none` to the end of the list, change it to `&kp`
4. Add a 2nd `&macro_param_1to1`
5. Add a binding (default is `&kp`)

And this is what's produced from `View keymap data`:
```
        macro_test: macro_test {
            compatible = "zmk,behavior-macro-one-param";
            #binding-cells = <1>;
            bindings = <&macro_param_1to1 &kp MACRO_PLACEHOLDER &macro_param_1to1 &kp>;
            label = "MACRO_TEST";
        };
```
which is missing the second `MACRO_PLACEHOLDER` at the end of the `bindings =` line.

nickcoutsos, 2023-11-22:
Ah, that's not great. I'll have to get back to take a deeper look later but I can reproduce it easily so I don't expect it'll be too much trouble. Thanks so much for the heads up!

nickcoutsos, 2023-11-23:
So the simplest summary is that it only injects the `MACRO_PLACEHOLDER` parameter on the first instance of any single `&macro_parameter_XtoY` directive. The oversight is caused by reusing the same code I to determine the parameter types. In that context I only want to consider the first instance, but it's still valid to reuse one parameter multiple times.

z5huang, 2023-11-23:
Thanks! Multiple `&macro_parameter_1to1` are used for example if you want to parametrize both the `&macro_press` and `&macro_release` (with e.g. a mod key)

While you are at it, there is another bug that may be related: if you first add two (non-parametric) bindings (e.g., `&kp A` and `&kp B`), then add two `&macro_parameter_1to1`, then move `&kp A` and `&kp B` each after their respective `&macro_parameter_1to1`, then the code generated is
```
        new_macro: new_macro {
            compatible = "zmk,behavior-macro-one-param";
            #binding-cells = <1>;
            bindings = <&macro_param_1to1 &kp MACRO_PLACEHOLDER &macro_param_1to1 &kp B>;
            label = "NEW_MACRO";
        };
```
where the trailing `B` in the `bindings =` line should not have been there.

nickcoutsos, 2023-11-23:
Yeah, so there's two pieces I think:

1. Changing a `&none` binding to `&kp` will normally require you to add the keycode parameter, except in the context of a macro where it's parameterized in which case I _think_ (haven't studied that code for a while now) it'll inject it for you, _but_ adding a `&kp` binding through the _Add Key Binding_ button will add it to the array without enforcing the same restriction or applying the same side effects
2. When serializing macros it deals with convenience bindings like `RGB_COLOR_HSB(...)` and injects `MACRO_PLACEHOLDER`s where appropriate, except in this case where its the second instance of that parameter being applied.

The other issue you've raised is a result of point number 2, where it's being overlooked because that macro parameter has been dealt with already. It's not the end of the world, because the actual value is meaningless, but it's inconsistent/unexpected and it's indicative of the larger issue that the logic isn't applying in all cases.

I'll need to look at how I can do this cleanly without breaking the parameter type resolution as it exists or duplicating parts of it but it shouldn't be that bad. If I have some quiet time tomorrow evening that should be enough.

nickcoutsos, 2023-11-29:
Sorry, I haven't forgotten about this I just haven't been able to concentrate on solving it. I have the code changes in place but I want to write a quality test or two before deploying anything.

closed 2023-12-03 by nickcoutsos