nickcoutsos/keymap-editor #159: Bug: error in `FileSystemPicker` when choosing  File System source

bendavis78, 2023-11-28:
When choosing local file system source, I see this error in the js console:

```
FileSystemPicker.d52…3e786ac7d9a678.js:2 Uncaught (in promise) TypeError: Cannot read properties of undefined (reading 'map')
    at P (FileSystemPicker.d52…ac7d9a678.js:2:3358)
P	@	FileSystemPicker.d52…3e786ac7d9a678.js:2
Promise.then (async)		
me	@	FileSystemPicker.d52…3e786ac7d9a678.js:2
a	@	FileSystemPicker.d52…3e786ac7d9a678.js:2
(anonymous)	@	FileSystemPicker.d52…3e786ac7d9a678.js:2
(anonymous)	@	FileSystemPicker.d52…3e786ac7d9a678.js:2
(anonymous)	@	FileSystemPicker.d52…3e786ac7d9a678.js:2
ni	@	react-dom.production.min.js:242
Ss	@	react-dom.production.min.js:285
(anonymous)	@	react-dom.production.min.js:281
B	@	vendors.fc16238….js:2
S	@	vendors.fc16238….js:2
```

nickcoutsos, 2023-11-28:
Oh, yeah, I really goofed that. I should be able to push a fix up in a couple of minutes!

nickcoutsos, 2023-12-03:
Hey Ben, I'm going to consider this resolved but if it turns out to not fix your case feel free to re-open.