nickcoutsos/keymap-editor #162: custom defines don't show up

nimishgautam, 2023-12-09:
I have custom defines in my keymap... things like:

`#define SCREENSHOT      LG(LS(N4))`

and later, in the layout I'll write:

`&kp SCREENSHOT`

This makes it much easier to read and understand later. Unfortunately the editor seems to have an error when it tries to parse these

nickcoutsos, 2023-12-09:
I get that it can simplify things in your keymap, but parsing preprocessor directives adds way too much complexity. If you're interested in details you can read the [preprocessor support wiki article](https://github.com/nickcoutsos/keymap-editor/wiki/Preprocessor-Support).

nimishgautam, 2023-12-09:
Fair enough. Thank you for the response (and clearly the fact that you've dealt with this before 🙂).

As you say in your wiki, if you're a person using preprocessor directives, you don't need this tool to configure your keyboard. 

Great tool anyways, and even without preprocessor support it's helping me keep my sanity as I try to refactor my keyboard layout, so thank you for that 🙂

closed 2023-12-09 by nimishgautam