nickcoutsos/keymap-editor #166: Feature Request: Change Keyboard Language

foxgreek, 2023-12-29:
Hi Nik, first of all - thanks for your amazing editor, it really saves my life configuring my corne keyboard. 

Do you think it is possible to add different keyboard language support?
I'm currently struggling with the German keyboard layout and I have to look up the English keys every time.

So it would be nice, if the editor automatically translate the inserted key/key-kombination from German Keyboard layout to English Keyboard layout.

Wdyt, might it be a lot of work?

nickcoutsos, 2023-12-29:
Hi!

I think maybe in the longer term that's the kind of direction I might take for supporting different languages but I need input from the people who would be using these features to make sure it actually solves problems.

There is some level of unofficial support now; if you're using [`zmk-locale-generator`](https://github.com/joelspadin/zmk-locale-generator) and have `#include "keys_de.h"` or similar in your keymap the editor will load those symbols and make them available. During key mapping you should be able to search that list based on the key code (`DE_SHARP_S`/`DE_SZ`) as well as the rendered symbol (`ß`) as long as you're currently able to type that. This means the `.keymap` code can contain meaningful locale-specific keycodes and the graphical layout can render Unicode symbols.

One important note is that this isn't "baking" the language into your keyboard, it's expected that you'll configure your OS to do the input mapping as appropriate.

I hope this is helpful, let me know if you have any questions or run into problems. You mention "automatically translate the inserted key/key-kombination" so I'll just point out that the locale mapping stuff doesn't currently apply to the Quick Key Assign feature.

nickcoutsos, 2024-01-08:
Hi @foxgreek, did you have any luck with zmk-locale-generator? Did I misunderstand what you're looking for?

premell, 2024-01-10:
this did not seem to work for me, but i might just do something wrong. I added the auto generated headers for swedish keys_sv.h to my config folder and imported it.
[image]
and in the .keymap: #include <keys_sv.h>

but they are not shown: 
[image]

Futhermore i tried to add some definitions individually but this didnt work either:
#define SV_O_UMLAUT (ZMK_HID_USAGE(HID_USAGE_KEY, HID_USAGE_KEY_KEYBOARD_SEMICOLON_AND_COLON))
#define SV_A_RING (ZMK_HID_USAGE(HID_USAGE_KEY, HID_USAGE_KEY_KEYBOARD_LEFT_BRACKET_AND_LEFT_BRACE))
#define SV_A_UMLAUT (ZMK_HID_USAGE(HID_USAGE_KEY, HID_USAGE_KEY_KEYBOARD_APOSTROPHE_AND_QUOTE))

You can see that the definitions are uploaded succesfully to the editor, but it doesnt seem to work either way.
[image]

nickcoutsos, 2024-01-10:
@premell 

The header include `<keys_sv.h>` should be `"keys_sv.h" because it's a local path.

Defining keybinds like that isn't supported by keymap editor. They'll compile, but the app won't parse them to understand what they are. Do those keycodes not exist in the headers from zmk-locale-generator or was that just for troubleshooting?

premell, 2024-01-10:
a nice it works, thank you. The repo looks pretty intimidating but it was actually pretty smooth. 

For reference if someone else is looking at this thread: 

You dont have to download the program or generate your own header files. Most languages already have header files here: https://github.com/joelspadin/zmk-locale-generator/releases. Just open the project locally and copy the file you want to the config/ folder and add #include "keys_de.h" (with the name of the file) in the .keymap file. Then push the changes to github and reload the editor.

closed 2024-01-14 by nickcoutsos