nickcoutsos/keymap-editor #173: Bug: Modifier toggles are blocked when selecting an unmodifiable key character

mattlbeck, 2024-01-18:
Just a small bug I noticed:

Pick the character "A" for a key and toggle a modifier e.g. LSHIFT. Then Attempt to re-pick a non-modifiable character e.g. "~". The mod toggles become uninteractable but still in the same phase as you left them. You must select a modifiable character to untick the modifier. Otherwise you are left with something like LSHIFT(TILDE) on the key, which won't have an effect but is a bit of a cludge.

Should the correct behaviour be to automatically untick all modifiers before blocking them?

nickcoutsos, 2024-01-18:
Hey, thanks for catching this!

Unless I'm misunderstanding, only the modifier that's being applied implicitly is non-toggleable. Since <kbd>~</kbd> is sent (in a US QWERTY layout) as <kbd>LShift</kbd>+<kbd>`</kbd> that particular toggle is disabled.

You're not wrong that the shift modifier should be cleared for the sake of elminating redundancy, but that can/should happen in a controlled manner instead of removing all modifiers. It shouldn't be terribly complicated to fix this but I haven't looked at that in a little while now and would need to double check on how this affects keycodes pulled from zmk-locale-generator.

nickcoutsos, 2024-01-18:
Just pushing a small fix for this now, it strips implicit modifiers when selecting a keycode (any other active modifiers are preserved but you can still toggle those at will). This should work well for keycodes coming from zmk-locale-generator but I'm actually using a hardcoded list of "shifted" keycodes for the standard ZMK set. If you run into something that might have been missed by that list feel free to reopen or create a new issue.

Thanks again!

closed 2024-01-18 by nickcoutsos