nickcoutsos/keymap-editor #175: Builds failing even from new keymap/fork

JamesMowery, 2024-01-20:
Attempted to create my own Github repo for a Corne wireless keyboard with nice!nano v2. Even from the very first build this was failing. Still continues to fail after a few hours of playing around with my own keymap. This is the error I'm seeing.

`Run west build -s zmk/app -d "/tmp/tmp.ytW6hHu9nt" -b "nice_nano_v2" -- -DZMK_CONFIG=/__w/corne-kbd-web/corne-kbd-web/config -DSHIELD="corne_left" 
+ west build -s zmk/app -d /tmp/tmp.ytW6hHu9nt -b nice_nano_v2 -- -DZMK_CONFIG=/__w/corne-kbd-web/corne-kbd-web/config -DSHIELD=corne_left
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /__w/corne-kbd-web/corne-kbd-web/zmk/app
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /__w/corne-kbd-web/corne-kbd-web/zephyr/.cache
-- Zephyr version: 3.2.0 (/__w/corne-kbd-web/corne-kbd-web/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Adding /__w/corne-kbd-web/corne-kbd-web/zmk/app/boards/shields/corne
-- ZMK Config directory: /__w/corne-kbd-web/corne-kbd-web/config
-- Using keymap file: /__w/corne-kbd-web/corne-kbd-web/config/corne.keymap
-- Using keymap file: /__w/corne-kbd-web/corne-kbd-web/config/corne.keymap
-- Board: nice_nano_v2
-- Shield(s): corne_left
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.2 (/opt/zephyr-sdk-0.15.2)
-- Found toolchain: zephyr 0.15.2 (/opt/zephyr-sdk-0.15.2)
-- Found Dtc: /opt/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /__w/corne-kbd-web/corne-kbd-web/zmk/app/boards/arm/nice_nano/nice_nano_v2.dts
-- Found devicetree overlay: /__w/corne-kbd-web/corne-kbd-web/zmk/app/boards/shields/corne/corne_left.overlay
-- Found devicetree overlay: /__w/corne-kbd-web/corne-kbd-web/zmk/app/boards/shields/corne/boards/nice_nano_v2.overlay
-- Found devicetree overlay: /__w/corne-kbd-web/corne-kbd-web/config/corne.keymap
'label' is marked as deprecated in 'properties:' in /__w/corne-kbd-web/corne-kbd-web/zmk/app/dts/bindings/behaviors/zmk,behavior-hold-tap.yaml for node /behaviors/autoshift_hold_tap.
'label' is marked as deprecated in 'properties:' in /__w/corne-kbd-web/corne-kbd-web/zmk/app/dts/bindings/behaviors/zmk,behavior-hold-tap.yaml for node /behaviors/homewrow_mods_left.
'label' is marked as deprecated in 'properties:' in /__w/corne-kbd-web/corne-kbd-web/zmk/app/dts/bindings/behaviors/zmk,behavior-hold-tap.yaml for node /behaviors/homewrow_mods_right.
'label' is marked as deprecated in 'properties:' in /__w/corne-kbd-web/corne-kbd-web/zmk/app/dts/bindings/behaviors/zmk,behavior-tap-dance.yaml for node /behaviors/capdance.
'label' is marked as deprecated in 'properties:' in /__w/corne-kbd-web/corne-kbd-web/zmk/app/dts/bindings/behaviors/zmk,behavior-sensor-rotate.yaml for node /behaviors/volume_encoder.
'label' is marked as deprecated in 'properties:' in /__w/corne-kbd-web/corne-kbd-web/zmk/app/dts/bindings/behaviors/zmk,behavior-sensor-rotate-var.yaml for node /behaviors/rgb_encoder.
'label' is marked as deprecated in 'properties:' in /__w/corne-kbd-web/corne-kbd-web/zmk/app/dts/bindings/behaviors/zmk,behavior-macro.yaml for node /macros/greeting.
'label' is marked as deprecated in 'properties:' in /__w/corne-kbd-web/corne-kbd-web/zmk/app/dts/bindings/behaviors/zmk,behavior-macro-two-param.yaml for node /macros/rgb_layer.
'label' is marked as deprecated in 'properties:' in /__w/corne-kbd-web/corne-kbd-web/zmk/app/dts/bindings/behaviors/zmk,behavior-macro-one-param.yaml for node /macros/with_lshift.
'label' is marked as deprecated in 'properties:' in /__w/corne-kbd-web/corne-kbd-web/zmk/app/dts/bindings/behaviors/zmk,behavior-macro-one-param.yaml for node /macros/autoshift.
devicetree error: <Node /pin-controller/i2c0_default in '/__w/corne-kbd-web/corne-kbd-web/zephyr/misc/empty_file.c'> lacks #sensor-binding-cells
CMake Error at /__w/corne-kbd-web/corne-kbd-web/zephyr/cmake/modules/dts.cmake:230 (message):
-- In: /tmp/tmp.ytW6hHu9nt/zephyr, command: /usr/bin/python3.8;/__w/corne-kbd-web/corne-kbd-web/zephyr/scripts/dts/gen_defines.py;--dts;/tmp/tmp.ytW6hHu9nt/zephyr/zephyr.dts.pre;--dtc-flags;'';--bindings-dirs;/__w/corne-kbd-web/corne-kbd-web/zmk/app/module/dts/bindings;/__w/corne-kbd-web/corne-kbd-web/zmk/app/dts/bindings;/__w/corne-kbd-web/corne-kbd-web/zephyr/dts/bindings;--header-out;/tmp/tmp.ytW6hHu9nt/zephyr/include/generated/devicetree_generated.h.new;--dts-out;/tmp/tmp.ytW6hHu9nt/zephyr/zephyr.dts.new;--edt-pickle-out;/tmp/tmp.ytW6hHu9nt/zephyr/edt.pickle;--vendor-prefixes;/__w/corne-kbd-web/corne-kbd-web/zmk/app/dts/bindings/vendor-prefixes.txt;--vendor-prefixes;/__w/corne-kbd-web/corne-kbd-web/zephyr/dts/bindings/vendor-prefixes.txt
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  /__w/corne-kbd-web/corne-kbd-web/zephyr/cmake/modules/zephyr_default.cmake:108 (include)
  /__w/corne-kbd-web/corne-kbd-web/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /__w/corne-kbd-web/corne-kbd-web/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:8 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/tmp/tmp.ytW6hHu9nt -GNinja -DBOARD=nice_nano_v2 -DZMK_CONFIG=/__w/corne-kbd-web/corne-kbd-web/config -DSHIELD=corne_left -S/__w/corne-kbd-web/corne-kbd-web/zmk/app
Error: Process completed with exit code 1.`

nickcoutsos, 2024-01-20:
The demo is a fake keyboard, corne doesn't actually have encoders.

JamesMowery, 2024-01-20:
Thanks! That fixed it. I feel like this web editor is going to be a big attraction to completely new users (like me). I think it would be great to assume that the user doesn't know what's going on (like this sensor thing you mentioned above, which I still don't understand). But I did see a sensor section on the first page and clicked delete. I still don't know what or why this does, but if I was a developer and wanted to gain adoption... please make the brand new user experience as easy as possible. Thanks!

JamesMowery, 2024-01-20:
> Edit: is there a particular keyboard you want to build keymaps for? You can use ZMK's setup script to create a repo for a corne or several other keyboards and for the most part the layout metadata is built into the app already.

Just seen this. I actually don't yet have the Corne here. So I haven't really gotten into that part (I think a lot of people who are just getting into this will probably do that after they build/receive their keyboard). I was just using the web editor to simulate what I think I want a layout to look like. Thus the note about this perhaps being an entryway for people who are completely new to this.

But good to know that the official ZMK stuff creates a repo for you.

I guess a good follow question is... should I NOT create a repo from the web editor at all and will all my work being done now be worthless? Or can I just use the repo (now fixed) for my Corne when it arrives and is built?

P.S. I really do appreciate this web editor! Just want it to not dissuade anyone from getting into this!

nickcoutsos, 2024-01-21:
You'll be able to migrate pretty easily without losing any of your work so far. What I'd recommend is following ZMK's steps to create a zmk-config repo set up for `corne` (or fork from your keyboard's vendor if they provide a more specific setup), and then copy your current `corne.keymap` into that repo's `config/` directory, leaving out the `sensor-bindings` properties.

closed 2024-01-28 by JamesMowery