nickcoutsos/keymap-editor #176: "Could not parse keymap" on brand new repo

BoweFlex, 2024-01-22:
Hopefully this isn't a common isssue, but after a good 30 seconds of looking I haven't found the answer.

Saw a video on your keymap editor and was really excited by how much easier this seemed to make the normal ZMK experience. I went through the steps from ZMK's install documentation (empty repo, run bash command, answer questions) and I got a stock keymap for my board (Sweep 34 key) that successfully builds in the github actions and provides me with the firmware I need. However, when I open Keymap editor and connect to that repo, I receive this error message:

``` 
It looks like some bindings in your keymap are using custom pre-processor macros which isn't supported by the Keymap Editor. These macros make it difficult both to resolve bindings to behaviors and parameters, as well as to generate modified keymap code that is consistent with your original intentions.

Please either continue editing your keymap by hand with these helpers or if you would like to use this editor, rewrite the keymap to use ZMK's syntax directly.

If you are looking for help in getting up and running or want a better understanding of the issue, please feel free to reach out to me:

    Keymap Editor Issues
    Discord - nickcoutsos#4351
```

I'd be surprised if this keymap is not valid, since it's the stock keymap provided by ZMK. Is there a bug with the editor, or is there something that I need to change to be able to use it? This is all pretty new to me so without spending a bunch of time digging in I'm not sure where to look to figure out what the difference is that this error is upset about. Thanks for your help!!

Github repo: https://github.com/BoweFlex/sweep-config

nickcoutsos, 2024-01-22:
Hi!

It's not _super_ common but some of the stock ZMK keymaps use preprocessor directives to achieve different things, and the keymap editor can't/won't parse this. "Invalid" wouldn't be the right way to describe it, as the keymap will of course compile, but right now the only way to change a keymap is to change the source code and there are some constraints on changes that this app will attempt.

For the details you can read [Keymap Editor Wiki: Preprocessor Support](https://github.com/nickcoutsos/keymap-editor/wiki/Preprocessor-Support), but in your case you can just change the problem on line 42 and carry on:

```patch
-           HRML(A,        S,         D,         F)     &kp G          &kp H      HRMR(J,        K,         L,        SQT)
+            &kp A     &kp S      &kp D      &kp F      &kp G          &kp H       &kp J     &kp K      &kp L     &kp SQT
```

The function call removed here was applying homerow modifiers to those bindings so if you'd like to reproduce the same functionality without manually applying it in the editor you can instead change the line to:

```patch
-           HRML(A,        S,         D,         F)     &kp G          &kp H      HRMR(J,        K,         L,        SQT)
+            &ht LSHFT A     &ht LALT S      &ht LCTRL D      &ht LGUI F      &kp G          &kp H       &ht RGUI J     &ht RCTRL K      &ht RALT L     &ht RSHFT SQT
```

BoweFlex, 2024-01-22:
I missed that wiki page - that makes sense. That fixed it, thank you for your help!!

closed 2024-01-22 by nickcoutsos