nickcoutsos/keymap-editor #177: Rotary encoder bindingsnot working

carlos-kassab-sp, 2024-01-22:
I've made sure my rotary encoder is working properly, the push button configuration works, but when I set up the rotation sensor it does not work.   For some reaso, the sensor bindings are not being picked up. I'm on a Sofle

Here's my repo https://github.com/KassabNova/zmk-config

nickcoutsos, 2024-01-22:
Oh, hey, you have to uncomment these two lines:

https://github.com/KassabNova/zmk-config/blob/2b6a9c3bf5f0537274be20b21d0739e7ac5afa47/config/sofle.conf#L8-L9

_(Make sure you're not just removing the `#` character, the space before `CONFIG_` needs to be removed as well)_

carlos-kassab-sp, 2024-01-22:
Oh man, I was going insane. Ty so much.
Your tool is really cool, is there a way to donate a coffee as a thank you?

closed 2024-01-22 by carlos-kassab-sp