nickcoutsos/keymap-editor #179: Syntax Error when defining combos

momueh, 2024-01-27:
Hi there! Just letting you know I ran into issues with the build for my corne when using combos.

Apparently there is a syntax error in the generated keymap file, and combos needs to be a separate node like so:

```
/ {
    combos {
        compatible = "zmk,combos";
        combo-esc {
             timeout-ms = <50>;
            bindings = <&kp ESC>;
            key-positions = <19 20>;
            layers = <0>;
        };
    };
};

/ {
    keymap {
    ...
};
```

In the file I copied over from the webapp it looked like:

```
/ {
    combos {
        compatible = "zmk,combos";
        combo-esc {
             timeout-ms = <50>;
            bindings = <&kp ESC>;
            key-positions = <19 20>;
            layers = <0>;
        };
    };

    keymap {
    ...
};
```

After changing this my build completed successfully.

nickcoutsos, 2024-01-27:
Hi!

Unless I'm overlooking a small detail the two examples you're showing are equivalent. The devicetree compiler (I think) in ZMK's build process will merge multiple root nodes into one -- my preference is to have one root node in your keymap file for simplicity and because it's just weird parsing it otherwise.

Do you have specific build errors you can share? I'd love to get to the bottom of this and make sure I'm not missing anything critical.

momueh, 2024-01-28:
Ok I checked again and you are correct, the syntax is not the issue here.

The reason for the issue seems to be that `#include <behaviors.dtsi>` is missing from the copied keymap file. I only added this line and then the build worked.

This is the minimal way I can reproduce the error:

1. Select Source "Clipboard" and Keyboard "Corne"
2. Add a combo
3. Copy keymap file

An error occurs in the Step "West Build": 
<details><summary>West Build output</summary>
<p>

```
Run west build -s zmk/app -d "/tmp/tmp.HS8MiyTFMP" -b "nice_nano_v2" -- -DZMK_CONFIG=/tmp/tmp.8SRheBqGpx/config -DSHIELD="corne_left" -DZMK_EXTRA_MODULES='/__w/zmk-config-corne/zmk-config-corne' 
  west build -s zmk/app -d "/tmp/tmp.HS8MiyTFMP" -b "nice_nano_v[2](https://github.com/momueh/zmk-config-corne/actions/runs/7684513123/job/20941094326#step:11:2)" -- -DZMK_CONFIG=/tmp/tmp.8SRheBqGpx/config -DSHIELD="corne_left" -DZMK_EXTRA_MODULES='/__w/zmk-config-corne/zmk-config-corne' 
  shell: sh -x {0}
  env:
    build_dir: /tmp/tmp.HS8MiyTFMP
    base_dir: /tmp/tmp.8SRheBqGpx
    zephyr_version: [3](https://github.com/momueh/zmk-config-corne/actions/runs/7684513123/job/20941094326#step:11:3).2.0
    extra_cmake_args: -DSHIELD="corne_left" -DZMK_EXTRA_MODULES='/__w/zmk-config-corne/zmk-config-corne'
    display_name: corne_left - nice_nano_v2
    artifact_name: corne_left-nice_nano_v2-zmk
+ west build -s zmk/app -d /tmp/tmp.HS8MiyTFMP -b nice_nano_v2 -- -DZMK_CONFIG=/tmp/tmp.8SRheBqGpx/config -DSHIELD=corne_left -DZMK_EXTRA_MODULES=/__w/zmk-config-corne/zmk-config-corne
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /tmp/tmp.8SRheBqGpx/zmk/app
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /tmp/tmp.8SRheBqGpx/zephyr/.cache
-- Zephyr version: 3.2.0 (/tmp/tmp.8SRheBqGpx/zephyr)
-- Found west (found suitable version "0.1[4](https://github.com/momueh/zmk-config-corne/actions/runs/7684513123/job/20941094326#step:11:4).0", minimum required is "0.7.1")
-- Adding /tmp/tmp.8SRheBqGpx/zmk/app/boards/shields/corne
-- ZMK Config directory: /tmp/tmp.8SRheBqGpx/config
-- ZMK Config Kconfig: /tmp/tmp.8SRheBqGpx/config/corne.conf
-- Using keymap file: /tmp/tmp.8SRheBqGpx/config/corne.keymap
-- Using keymap file: /tmp/tmp.8SRheBqGpx/config/corne.keymap
-- Board: nice_nano_v2
-- Shield(s): corne_left
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.1[5](https://github.com/momueh/zmk-config-corne/actions/runs/7684513123/job/20941094326#step:11:5).2 (/opt/zephyr-sdk-0.15.2)
-- Found toolchain: zephyr 0.15.2 (/opt/zephyr-sdk-0.15.2)
-- Found Dtc: /opt/zephyr-sdk-0.15.2/sysroots/x8[6](https://github.com/momueh/zmk-config-corne/actions/runs/7684513123/job/20941094326#step:11:6)_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /tmp/tmp.[8](https://github.com/momueh/zmk-config-corne/actions/runs/7684513123/job/20941094326#step:11:8)SRheBqGpx/zmk/app/boards/arm/nice_nano/nice_nano_v2.dts
-- Found devicetree overlay: /tmp/tmp.8SRheBqGpx/zmk/app/boards/shields/corne/corne_left.overlay
-- Found devicetree overlay: /tmp/tmp.8SRheBqGpx/zmk/app/boards/shields/corne/boards/nice_nano_v2.overlay
-- Found devicetree overlay: /tmp/tmp.8SRheBqGpx/config/corne.keymap
devicetree error: /combos/esc: undefined node label 'kp'
CMake Error at /tmp/tmp.8SRheBqGpx/zephyr/cmake/modules/dts.cmake:230 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  /tmp/tmp.8SRheBqGpx/zephyr/cmake/modules/zephyr_default.cmake:108 (include)
  /tmp/tmp.8SRheBqGpx/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /tmp/tmp.8SRheBqGpx/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:[9](https://github.com/momueh/zmk-config-corne/actions/runs/7684513123/job/20941094326#step:11:9)2 (include_boilerplate)
  CMakeLists.txt:8 (find_package)

-- In: /tmp/tmp.HS8MiyTFMP/zephyr, command: /usr/bin/python3.8;/tmp/tmp.8SRheBqGpx/zephyr/scripts/dts/gen_defines.py;--dts;/tmp/tmp.HS8MiyTFMP/zephyr/zephyr.dts.pre;--dtc-flags;'';--bindings-dirs;/tmp/tmp.8SRheBqGpx/zmk/app/module/dts/bindings;/tmp/tmp.8SRheBqGpx/zmk/app/dts/bindings;/tmp/tmp.8SRheBqGpx/zephyr/dts/bindings;--header-out;/tmp/tmp.HS8MiyTFMP/zephyr/include/generated/devicetree_generated.h.new;--dts-out;/tmp/tmp.HS8MiyTFMP/zephyr/zephyr.dts.new;--edt-pickle-out;/tmp/tmp.HS8MiyTFMP/zephyr/edt.pickle;--vendor-prefixes;/tmp/tmp.8SRheBqGpx/zmk/app/dts/bindings/vendor-prefixes.txt;--vendor-prefixes;/tmp/tmp.8SRheBqGpx/zephyr/dts/bindings/vendor-prefixes.txt
-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/tmp/tmp.HS8MiyTFMP -GNinja -DBOARD=nice_nano_v2 -DZMK_CONFIG=/tmp/tmp.8SRheBqGpx/config -DSHIELD=corne_left -DZMK_EXTRA_MODULES=/__w/zmk-config-corne/zmk-config-corne -S/tmp/tmp.8SRheBqGpx/zmk/app
Error: Process completed with exit code 1.
```

</p>
</details> 

<details><summary>copied keymap file</summary>
<p>

```
[... 41 lines cut ...]
&kp LSHFT   &trans        &trans        &trans        &trans        &trans          &trans    &trans    &trans    &trans     &trans  &trans
                                        &kp LGUI      &trans        &kp SPACE       &kp RET   &trans    &kp RALT
            >;
        };

        raise_layer {
[... cut ...]
&kp TAB    &kp EXCL  &kp AT  &kp HASH  &kp DLLR  &kp PRCNT    &kp CARET  &kp AMPS   &kp KP_MULTIPLY  &kp LPAR  &kp RPAR  &kp BSPC
&kp LCTRL  &trans    &trans  &trans    &trans    &trans       &kp MINUS  &kp EQUAL  &kp LBKT         &kp RBKT  &kp BSLH  &kp GRAVE
&kp LSHFT  &trans    &trans  &trans    &trans    &trans       &kp UNDER  &kp PLUS   &kp LBRC         &kp RBRC  &kp PIPE  &kp TILDE
                             &kp LGUI  &trans    &kp SPACE    &kp RET    &trans     &kp RALT
            >;
        };
    };
};
```

</p>
</details>

nickcoutsos, 2024-01-28:
That makes much more sense, thank you for clarifying! I should be able to make the change today -- ideally it should automatically inject the header if necessary like it does for  behavior-specific headers, but at the very least I can include it in the "bare minimum" keymap content.

Out of curiosity, did you already have an existing keymap with the ascii layer diagrams or did you add those afterwards?

nickcoutsos, 2024-01-28:
Ok, I'm now treating `behaviors.dtsi` as a "managed" include that gets injected regardless of used behaviors.

Thanks for bringing up the issue!

closed 2024-01-28 by nickcoutsos