nickcoutsos/keymap-editor #183: Deleting top Reconfiguration has no effect

MikeSE314, 2024-02-08:
## Steps to reproduce

Using the Keymap editor in clipboard mode, load in an example keymap, such as A. Dux. Navigate to the Behaviors section. Under Reconfigurations, select Add Behavior Reconfiguration, add any behavior binding, such as &caps_words. Click Create, then Okay. Click the Keymap button to confirm changes.

```
#include <behaviors.dtsi>

&caps_word {
};

/ {
    keymap { compatible = "zmk,keymap"; };
};
```

Delete and confirm delete the reconfiguration. Then click the Keymap button again. Note the contents haven't changed, and the reconfiguration has reappeared.

nickcoutsos, 2024-02-08:
Interesting!

I can reproduce it in minimal keymaps when a such a reconfigured behavior is the first syntax node following a pre-processor directive so I think this has to do with some of the goofiness surrounding handling of newlines.

I'll start digging into that and will keep you posted. To get an idea of urgency, does that empty `&caps_word {};` node cause build/firmware problems or is it just an annoyance right now? I guess in the worst case you can manually remove it but yeah I fully agree this needs to be resolved.

MikeSE314, 2024-02-08:
Totally just an annoyance. I had to remove mine by hand. Thanks for looking into it!

nickcoutsos, 2024-02-08:
Found the issue! In cases where I'm deleting the last node of a collection but it doesn't have a defined "container" like combos, keymap layers, or conditional layers it can end up not flushing the changes to the devicetree document. This could technically have presented itself with macros/behaviors but by convention people tend to place these in a `macros {};` or `behaviors {};` node.

Anyway, this fix is being deployed now. Thanks for reporting this!

closed 2024-02-08 by nickcoutsos