nickcoutsos/keymap-editor #188: Feature Request: Print Multiple layers in one cheatsheet

m42e, 2024-02-24:
It would be Nice to be able to print the layout with multiple key bindings on one page, so each key would look like that:

```
1   2
  0  
3   4
```

So you can easily use it as cheatsheet while learning a new layout.

nickcoutsos, 2024-02-24:
Printing/exporting your keymap in a diagram is a significant feature on its own (see https://github.com/nickcoutsos/keymap-editor/discussions/141) and I'd already started some preliminary work there, but I'm sorry to say that it almost certainly wouldn't support what you're looking for.

I think the simplest explanation is that doing that programmatically is difficult. Usually when you see such cheatsheets people will customize it with shorthands and legends because the density of information is much higher and someone/something needs to decide which details are most important to keep things readable and useful.

Once you stray out of simple keycodes the context of the behavior matters (is the `1` a `&mo` binding or a `&tog`, or a `&bt BT_SEL`?) and I think that's the kind of thing people might make a judgement call to omit because they're familiar enough with their own keymap that they'll be able to infer that missing context easily.

If you *do* want to have that kind of customization you might be able to get it with [keymap-drawer](https://github.com/caksoylar/keymap-drawer/). I haven't done it myself but I know there's a lot of flexibility with custom CSS classes.

m42e, 2024-02-24:
Thanks for your explanation. Thanks for all the effort spent. I can see the issues. 

And thanks again for all the effort spent on this project!

closed 2024-02-24 by m42e