nickcoutsos/keymap-editor #190: right encoder missing

yundddd, 2024-03-02:
For some reason, only encoder_left shows up in the configuration. I have configured sensor bindings as:
`sensor-bindings = <&inc_dec_kp RIGHT LEFT &inc_dec_kp UP DOWN>;`
Is there anything else I need to configure?

config file: https://github.com/yundddd/zmk_test_lily58/edit/lily58/config/lily58.keymap

Thanks!

nickcoutsos, 2024-03-03:
Put this in `config/lily58.json`:

```json
{
  "id": "lily58",
  "name": "Lily58",
  "layouts": {
    "default_transform": {
      "name": "default_transform",
      "layout": [
        { "row": 0, "col":  0, "x":     1, "y":  0.4 },
        { "row": 0, "col":  1, "x":     2, "y": 0.33 },
        { "row": 0, "col":  2, "x":     3, "y": 0.07 },
        { "row": 0, "col":  3, "x":     4, "y":    0 },
        { "row": 0, "col":  4, "x":     5, "y": 0.07 },
        { "row": 0, "col":  5, "x":     6, "y":  0.2 },
        { "row": 0, "col":  9, "x":    10, "y":  0.2 },
        { "row": 0, "col": 10, "x":    11, "y": 0.07 },
        { "row": 0, "col": 11, "x":    12, "y":    0 },
        { "row": 0, "col": 12, "x":    13, "y": 0.07 },
        { "row": 0, "col": 13, "x":    14, "y": 0.33 },
        { "row": 0, "col": 14, "x":    15, "y":  0.4 },

        { "row": 1, "col":  0, "x":     1, "y":  1.4 },
        { "row": 1, "col":  1, "x":     2, "y": 1.32 },
        { "row": 1, "col":  2, "x":     3, "y": 1.07 },
        { "row": 1, "col":  3, "x":     4, "y":    1 },
        { "row": 1, "col":  4, "x":     5, "y": 1.08 },
        { "row": 1, "col":  5, "x":     6, "y":  1.2 },
        { "row": 1, "col":  9, "x":    10, "y":  1.2 },
        { "row": 1, "col": 10, "x":    11, "y": 1.08 },
        { "row": 1, "col": 11, "x":    12, "y":    1 },
        { "row": 1, "col": 12, "x":    13, "y": 1.07 },
        { "row": 1, "col": 13, "x":    14, "y": 1.32 },
        { "row": 1, "col": 14, "x":    15, "y":  1.4 },

        { "row": 2, "col":  0, "x":     1, "y": 2.41 },
        { "row": 2, "col":  1, "x":     2, "y": 2.33 },
        { "row": 2, "col":  2, "x":     3, "y": 2.08 },
        { "row": 2, "col":  3, "x":     4, "y": 2.01 },
        { "row": 2, "col":  4, "x":     5, "y": 2.08 },
        { "row": 2, "col":  5, "x":     6, "y":  2.2 },
        { "row": 2, "col":  9, "x":    10, "y":  2.2 },
        { "row": 2, "col": 10, "x":    11, "y": 2.08 },
        { "row": 2, "col": 11, "x":    12, "y": 2.01 },
        { "row": 2, "col": 12, "x":    13, "y": 2.08 },
        { "row": 2, "col": 13, "x":    14, "y": 2.33 },
        { "row": 2, "col": 14, "x":    15, "y": 2.41 },

        { "row": 3, "col":  0, "x":     1, "y": 3.41 },
        { "row": 3, "col":  1, "x":     2, "y": 3.33 },
        { "row": 3, "col":  2, "x":     3, "y": 3.08 },
        { "row": 3, "col":  3, "x":     4, "y": 3.01 },
        { "row": 3, "col":  4, "x":     5, "y": 3.08 },
        { "row": 3, "col":  5, "x":     6, "y":  3.2 },
        { "row": 3, "col":  6, "x":     7, "y":  2.7 },
        { "row": 3, "col":  8, "x":     9, "y":  2.7 },
        { "row": 3, "col":  9, "x":    10, "y":  3.2 },
        { "row": 3, "col": 10, "x":    11, "y": 3.08 },
        { "row": 3, "col": 11, "x":    12, "y": 3.01 },
        { "row": 3, "col": 12, "x":    13, "y": 3.08 },
        { "row": 3, "col": 13, "x":    14, "y": 3.33 },
        { "row": 3, "col": 14, "x":    15, "y": 3.41 },

        { "row": 4, "col":  3, "x":  3.49, "y": 4.08 },
        { "row": 4, "col":  4, "x":  4.49, "y": 4.08 },
        { "row": 4, "col":  5, "x":  5.49, "y":  4.2 },
        { "row": 4, "col":  6, "x":  6.49, "y":  3.7, "h": 1.5, "r":  30, "rx":  6.49, "ry": 5.2 },
        { "row": 4, "col":  8, "x":  9.51, "y":  3.7, "h": 1.5, "r": -30, "rx": 10.51, "ry": 5.2 },
        { "row": 4, "col":  9, "x": 10.51, "y":  4.2 },
        { "row": 4, "col": 10, "x": 11.51, "y": 4.08 },
        { "row": 4, "col": 11, "x": 12.51, "y": 4.08 }
      ]
    }
  },
  "sensors": [
    {
      "ref": "left_encoder",
      "name": "encoder_left",
      "identifier": "encoder_left",
      "compatible": "alps,ec11",
      "label": "LEFT_ENCODER"
    },
    {
      "ref": "right_encoder",
      "name": "encoder_right",
      "identifier": "encoder_right",
      "compatible": "alps,ec11",
      "label": "RIGHT_ENCODER"
    }
  ]
}
```

yundddd, 2024-03-03:
Sweet that worked! One related question about the encoder, do you happen to know where to put these resolution blocks? I have tried adding them into the keymap but failed to compile.

https://github.com/zmkfirmware/zmk/pull/728#issuecomment-861932593

nickcoutsos, 2024-03-03:
Uh, huh. It looks like ZMK's lily58 shield only has the left encoder. You would need to add the other encoder device... it's a bit more complicated.

I _think_ you can add `config/lily58.dtsi` with:

```dts
    right_encoder: encoder_right {
        compatible = "alps,ec11";
        a-gpios = <&pro_micro 21 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP)>; // change these
        b-gpios = <&pro_micro 20 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP)>;
        resolution = <2>;
    };

    &left_encoder { resolution = <2>; }; // might as well throw this in here instead of the keymap

    &sensors {
      sensors = <&left_encoder &right_encoder>;
    }
```

That will define the right encoder (and set the correct resolution property for the left) and add it to the list of sensors.

And also `config/lily58_right.overlay`:

```dts
&right_encoder {
    status = "okay";
};
```

To enable the encoder on that side.

I _think_. I don't have anything with encoders and I haven't had to try and overlay things onto existing ZMK shield definitions.

nickcoutsos, 2024-03-10:
Hey, just wanted to follow up, I looked at your repo and see the added `lily58.dtsi` file. I forgot to structure the code properly; the right encoder node needs to be inside the root node, so the file would look like:

```dts
/ {
    right_encoder: encoder_right {
        compatible = "alps,ec11";
        a-gpios = <&pro_micro 21 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP)>; // change these
        b-gpios = <&pro_micro 20 (GPIO_ACTIVE_HIGH | GPIO_PULL_UP)>;
        resolution = <2>;
    };
};

&left_encoder { resolution = <2>; };

&sensors {
    sensors = <&left_encoder &right_encoder>;
}
```

yundddd, 2024-03-12:
`devicetree error: /__w/zmk_test_lily58/zmk_test_lily58/config/lily58_right.overlay:1 (column 1): parse error: undefined node label 'right_encoder'`

unfortunately I didn't get to look deep into this. I did manage to get sofle firmware onto lily 58 though by ignoring two extra keys. Both encoders work perfectly with the right resolution out of box. Not perfect but it works! Thanks for the suggestion though I will try to experiment with it when I get more time...

nickcoutsos, 2024-03-12:
I had another mistake (didn't have `#include "lily58.dtsi"`) but this config structure still has problems. I think the most appropriate thing would be editing the original lily58 config to add the right encoder there (and leave it disabled if need be) but maybe that's based on a specific version of the lily58 pcb that doesn't support it?

If your setup is working that's the most important thing so I'll leave it at that. We're past dealing with issues in the scope of keymap-editor so I'll close the issue now, but if you do run into other problems feel free to open a new one.

closed 2024-03-12 by nickcoutsos

GregoryDi, 2025-05-02:
Hi, I had exactly the same problem and after several tries I managed to add right encoder with mouse scroll behaviour in lily58, so I desided to send my build here, maybe it would be useful for people who have similar problems. Maybe some files are redundant, but it definitely works as expected.
https://github.com/GregoryDi/zmk-config