nickcoutsos/keymap-editor #191: Down?

goudacris, 2024-03-06:
Hangs when 'login with github' is clicked

nickcoutsos, 2024-03-06:
Yep, sorry for the inconvenience, I'm having some problems with the hosting platform. In the meantime you can use the _Clipboard_ source to make edits without the GitHub integration as it doesn't rely on the backend.

nickcoutsos, 2024-03-06:
I've just rebuilt the API's deploy configuration and you can now load keymaps from GitHub again.

If you're seeing something like this:

[image]

Just refresh the page.

64Dundy, 2024-03-06:
Thank you! When I save the changes I've made, they don't get applied. Most likely user error but do I need to turn my keyboard on/off or should it apply once saved successfully? I forked the repository as well. Please let me know if you happen to know!

nickcoutsos, 2024-03-06:
If I'm understanding you correctly -- Keymap Editor can't flash your keyboard firmware. All it's really doing is manipulating the code for your `.keymap` file and (in this case) committing those changes to your GitHub repo. From then on it's relying on GitHub's actions to automatically build the firmware and for you to flash the firmware to your keyboard.

64Dundy, 2024-03-06:
I see, I was under the impression this was a graphical keymap editor. I guess I will have to edit/configure the code in the .keymap file if I want to change the key mapping. Thank you for the assistance!

nickcoutsos, 2024-03-06:
That actually is the function of the editor. If you look at your repository the code is being updated when you save changes in the app. The app will also link to the latest build where you can download the compiled firmware and flash it to your keyboard.

64Dundy, 2024-03-06:
I understand now, I then view the code generated from the editor and then apply it to my keyboard? Thank you for explaining that I am very new to this. I am now researching how I can apply the keymap file to my keyboard. Thank you again

nickcoutsos, 2024-03-06:
When you load your keymap in the app you'll see something like this at the top:

[image]

The blue button shows the last build and clicking it will take you to the summary in GitHub. In your case right now that's here: https://github.com/64Dundy/lily58-wireless-view-zmk-config/actions/runs/8177104650

On that page under the _Artifacts_ section there's a link to download _firmware_ which is a zip file containing

- `lily58_left nice_view_adapter nice_view-nice_nano_v2-zmk.uf2`, and
- `lily58_right nice_view_adapter nice_view-nice_nano_v2-zmk.uf2`

which you'll flash to your left and right keyboard halves respectively. In most cases (certainly the ones where you're making changes through keymap editor) you can get away with only flashing the left side, but if you'd rather be consistent about it you can do both.

As for the flashing process, it's described here: https://docs.typeractive.xyz/build-guides/lily58-wireless/firmware#installing-firmware

64Dundy, 2024-03-06:
You are amazing thank you for the help again. It worked : ) using my new lily58 right now. Thank you! Have a great day

closed 2024-03-10 by nickcoutsos