nickcoutsos/keymap-editor #192: Fresh GH login doesn't seem to fully install/authorize?

petejohanson, 2024-03-08:
Seems like a fresh login on the editor is getting borked on the GH side of things:

[image]
[image]

Clicking the fix link goes to a 404, if I instead inspect my authorized/added apps, I only see this:

[image]
[image]

nickcoutsos, 2024-03-08:
This is really strange! It doesn't appear in the installed GitHub apps?

If you've just logged in there should be no installation so it should take you through that flow. In the past I've seen some weirdness regarding installations for users vs organizations and I wonder if that could be a factor.

When you hit the 404 page does the URL contain an installation id?

petejohanson, 2024-03-08:
> This is really strange! It doesn't appear in the installed GitHub apps?

Nope. Only authorized.

> 
> If you've just logged in there should be no installation so it should take you through that flow. In the past I've seen some weirdness regarding installations for users vs organizations and I wonder if that could be a factor.

Maybe? I am a member of a couple organizations.

> 
> When you hit the 404 page does the URL contain an installation id?

Sure does: https://github.com/settings/installations/28146131

nickcoutsos, 2024-03-08:
After some investigation it seems like the situation is:

- During login the app fetches a list of your installations (e.g. for your user account versus an organization account), if any
- GitHub's API returns installations which include repositories for which the authenticated user has explicit access (including explicit read permission, even if the repo is public)
- When the app gets a single suspended installation it won't prompt you to create a new installation, but rather to update that installation's repository selection
- If you only have read access to said repo you probably don't have admin access to the installation, and therefore won't have access to the settings page.

The end result is being locked out of the app and having no way to remedy the situation yourself.

Short term fix is filtering out installations you can't access. At some point I should update the flow to enable installing for another account if appropriate, that will need some more investigation.

nickcoutsos, 2024-03-09:
Some notes for future me:

- The `permissions` property of the returned installation object refers to what the installation allows the app to do to selected repositories
- Nothing in the installation object explicitly tells us whether the authenticated user can view/alter installation settings, but basically:
  - An installation with `account.type: "User"` and `account.login: <authenticated user>` is editable
  - An installation with `account.type: "User"` and `account.login: <anyone else>` is not
  - An installation with `account.type: "Organization"`, I don't know. There's an API to get organization membership for a specific user, that might be enough.

I think this means the UI necessarily needs to get a little more complicated. When getting the list of installations we need to know which are editable before including the "Add Repository" button or the "Manage repositories for [...]" in the repo picker.

closed 2024-03-10 by nickcoutsos

petejohanson, 2024-03-10:
> Last update: the actual surface level problem was that I was incorrectly generating the settings URL for app installations on organizations. To avoid the complexity of listing any organizations in the UI and distinguishing between their installations and the user's personal installation I'm now just linking to GitHub's app installations list so there's always the option of configuring an existing installation or creating a new one.

Thanks!