nickcoutsos/keymap-editor #193: Duplicate Layer

FabianoLothor, 2024-03-11:
I'd like to see this option available.

I probably could try and contribute to the project adding this feature if you want, it looks a simple and quick win.

nickcoutsos, 2024-03-11:
That's a fair point, I've recently(-ish) added support for duplicating other node types so it wouldn't be terribly difficult to add here. I can take a look later this afternoon, most likely it would involve a new option in the layer actions dropdown.

nickcoutsos, 2024-03-11:
Just pushed, have a look (refresh if you're still in the app) when you have a moment!

FabianoLothor, 2024-03-12:
Working great, much appreciated!

FabianoLothor, 2024-03-12:
Found a possible bug, hopefully you can reproduce by yourself:

1. Duplicate layer X.
2. Rename duplicated layer Y.
3. Save your changes.
  - I'm using github, unsure if happens with other sources.
4. Refresh the page or hit the refresh button.

Layer X is saved with the same name of layer Y.

That's likely happening due JS copying the reference of the previous object.

In the end, that's the result I'm getting:

[image]

nickcoutsos, 2024-03-12:
Ah shit!

My mind went to the reference thing, too, but I'm actually deep-cloning the layer. That's the problem, incidentally, because when the keymap is parsed I store an index for each node to help manage changes (add/remove/reorder) to the collection. I should just need to remove the index property of the cloned node but I'll want to wait until morning and do some more testing. It would also mean that any changes you make to the duplicated layer's bindings would also blow out the original which would be incredibly annoying.

Thanks for catching this!

nickcoutsos, 2024-03-12:
Should be resolved now! 🤞

FabianoLothor, 2024-03-12:
Thank you, couldn't reproduce anymore, closing it.

closed 2024-03-12 by FabianoLothor

nickcoutsos, 2024-03-12:
Fantastic, thanks again @FabianoLothor!