nickcoutsos/keymap-editor #197: Warning for combo variables

FabianoLothor, 2024-03-14:
I recently got an issue when creating multiple combos for the same key due the value of `CONFIG_ZMK_COMBO_MAX_COMBOS_PER_KEY`.

Its default value is 5: https://zmk.dev/docs/config/combos

I'd 7 macros configured for the same keys and it took me some time until figure out what was the issue.

Do you see any value in add some sort of warning when this happens?

We also have:
- `CONFIG_ZMK_COMBO_MAX_PRESSED_COMBOS` default 4;
- `CONFIG_ZMK_COMBO_MAX_KEYS_PER_COMBO` default 4;

Certainly this doesn't happen often and I would consider it just a "nice to have" feature.

nickcoutsos, 2024-03-14:
Definitely a nice to have. The trouble for me in the short term is that although we have these configurations to look at and they're all well documented, there are so many places where the configuration _could_ exist and _could_ change between fetches that it ends up being a lot of API calls.

Eventually this would be a non-issue because all of the data I need will come directly from the keyboard through ZMK Studio's protocol.

As a short term thing I might use the defaults as "typical" limits and present warnings in the UI that make it clear individual users may have different setups, linking to relevant documentation for additional details.

ujl123, 2024-07-08:
Came in here by searching "CONFIG_ZMK_COMBO_MAX_COMBOS_PER_KEY" and had the same issue. There was no warning regarding going over the max default and I tried everything like debugging key positions, trying different key codes etc until I worked it out.  A soft warning until the new api would be cool. 

That being said, when does the new ZMK studio protocols start?

nickcoutsos, 2024-07-10:
Preliminary work on ZMK Studio is already started, but this project is unrelated. Sorry I left this hanging so long. I'll have to see where I left off with trying to lay out the warning messages and see if there's a less involved approach for the short term.