nickcoutsos/keymap-editor #198: Local includes don't show up in keymap editor

JohnnyMcWeed, 2024-03-18:
I have a [keymap](https://github.com/JohnnyMcWeed/keyboard/blob/myconf/config/adv360.keymap) where  I added some custom local includes.
```
#include "keymap_swiss_de.h"
```
```
/ {
    behaviors {
        #include "macros.dtsi"
   }
}
```

[Build works](https://github.com/JohnnyMcWeed/keyboard/actions), I just get errors/warnings in the web editor (`undefined` / `binding has missing values`) when viewing my config and I guess I wouldn't be able to use it as those macros do not show up.

nickcoutsos, 2024-03-18:
Includes aren't processed for a couple of reasons:

1. I try to limit the number of requests that the backend needs to make
2. Loading included files could mean arbitrary headers for:
    - Aliased keycodes/behaviors via [preprocessor macros](https://github.com/nickcoutsos/keymap-editor/wiki/Preprocessor-Support) which I'd ignore, or
    - Custom behaviors/combos/etc, which I could parse and edit, but is it appropriate to merge those changes to the original file? Probably, but what about additions?

The exception to this is headers for [zmk-locale-generator](https://github.com/joelspadin/zmk-locale-generator), although I'm not actually fetching the header. I've pulled out the symbols and aliases from those headers and they become available for selection/rendering when a matching filename is encountered.

JohnnyMcWeed, 2024-03-21:
Thank you very much for the clarification.

closed 2024-03-21 by JohnnyMcWeed

nickcoutsos, 2024-03-21:
Just to clarify a couple of points:

1. The app will recognize include statements for zml-locale-generator headers but it won't add them itself, so if you're including `"keys_XXX.h"` it's still up to you to make sure that file exists in your repo.
2. If you're including custom behaviors in other files they're not parsed by the app either, so it'll probably complain about your use of unrecognized behaviors

And lastly `#ifdef`/`#ifndef`/`#endif` aren't handled well, so if you intend to make and save changes through keymap editor you'll want to remove those first.