nickcoutsos/keymap-editor #200: Glove 80 action build failed

websorz, 2024-03-20:
Hello, I am having issues getting the web tool to work with my glove80. The build fails after adding a single macro to the layout with the following : 

`error: 1 dependencies of derivation '/nix/store/b6cb04ii6maidv143pakqim69pw76mbn-combined_zmk_glove80_lh_zmk_glove80_rh.drv' failed to build` 

here is the action build:  https://github.com/websorz/glove80_config/actions/runs/8364672469/job/22901099219

nickcoutsos, 2024-03-20:
Hmm... Keymap Editor really isn't going to play well with all the [preprocessor stuff](https://github.com/nickcoutsos/keymap-editor/wiki/Preprocessor-Support). In this case it looks like things start following apart when it incorrectly parsed/formatted the `#endif` on line 31.

If you're motivated to use Keymap Editor I'd recommend removing those preprocessor conditionals. It doesn't look like `#ifndef LAYER_Lower` is relevant here, and `#ifdef BT_DISC_CMD` shouldn't be needed if you're building against ZMK's main branch. I can give you more specific pointers there if you'd like.

nickcoutsos, 2024-03-25:
I'll close this, but let me know if you'd like help adapting things.

closed 2024-03-25 by nickcoutsos