nickcoutsos/keymap-editor #201: Feature request, add function to copy key config

R3PTR, 2024-03-21:
Id like a function to copy key, would be helpful for keys like &none, for example, i have a corne with 5 instead of 6 rows and need the outer rows to be none. It is quite tedious to do all manually

nickcoutsos, 2024-03-21:
Hi @R3PTR 

There's been a request for this already and it's definitely worth doing, but in your case you'd be better off using the five column matrix transform. If your keymap file includes

```dts
/ {
  chosen {
    zmk,matrix-transform = &five_column_transform;
  };
};
```

then Keymap Editor will use the appropriate layout and you won't need the redundant `&none` binds in your keymap bindings arrays.

R3PTR, 2024-03-21:
Ah, thank you for the info.

ArturZurawski, 2024-11-26:
this config prevents `CONFIG_ZMK_STUDIO=y` from working. I tried
```
/ {
    chosen {
        zmk,physical-layout = &foostan_corne_5col_layout;
    };
};
```

but this does not change layout in keymap-editor. Are there any alternatives?

nickcoutsos, 2024-11-27:
I haven't been able to keep up with the ZMK Studio development, you're saying it fails to build when using `zmk,matrix-transform` instead?

That's unfortunate, but I suspect you'll run into other incompatibilities as well. I was looking into things earlier on but stopped due to the depth of physical layouts. There's also things like having a fixed number of layers/behaviors/etc that I'm not accounting for, and just in general I don't think there's going to be an easy path to dealing with both an editable devicetree keymap and runtime configurations -- you might just need to pick one or the other.

ArturZurawski, 2024-11-27:
Unfortunately [no](https://zmk.dev/docs/features/studio#adding-zmk-studio-support-to-a-keyboard)
> The keyboard should also not have a chosen zmk,matrix-transform

I disabled zmk studio for now. Your editor is better :)