nickcoutsos/keymap-editor #209: &soft_off not an option

grassfedreeve, 2024-04-07:
&soft_off has been merged into ZMK main, would be great to see this included in Keymap Editor!

https://zmk.dev/docs/features/soft-off
https://zmk.dev/docs/behaviors/soft-off
https://zmk.dev/docs/config/power#soft-off

nickcoutsos, 2024-04-07:
Hey, thanks for the heads up. I've updated the set of behaviors but I'm having trouble with deploys right now which affects keymaps loading from GitHub. If you're using another source like Clipboard/Filesystem you should have the new behavior available.

I'll update here again when I'm able to update the backend.

nickcoutsos, 2024-04-09:
I just finished migrating my API backend to a new platform so this is now complete. If you run into any trouble with the behavior please re-open or create a new issue. Thanks!

closed 2024-04-09 by nickcoutsos