nickcoutsos/keymap-editor #210: Parse error on generated keymap

AhmedAA, 2024-04-07:
Hello!

Thank you for this amazing tool, it really saved me a lot of time in terms of manual typing, appreciate it.

I generated a keymap file for my Sofle board, but for some weird reason it is not able to parse it.
The zmk github action fails on line 44, column 53, which in my head makes no sense. I hope someone can help me figure it out.

The error message: "devicetree error: /tmp/zmk-config/config/sofle.keymap:44 (column 53): parse error: expected number or parenthesized expression"

The resulting keymap is posted below:
```
#include <behaviors.dtsi>
#include <dt-bindings/zmk/keys.h>
#include <dt-bindings/zmk/bt.h>
#include <dt-bindings/zmk/rgb.h>
#include <dt-bindings/zmk/ext_power.h>

/ {
    keymap {
        compatible = "zmk,keymap";

        base {
            bindings = <
&trans  &kp N1      &kp N2           &kp N3       &kp N4        &kp N5                                  &kp N6           &kp N7       &kp N8       &kp N9             &kp N0                 &trans
&trans  &kp Q       &kp W            &kp E        &kp R         &kp T                                   &kp Y            &kp U        &kp I        &kp O              &kp P                  &trans
&trans  &mt LGUI A  &mt LALT S       &mt LCTRL D  &mt LSHFT F   &kp G                                   &kp H            &mt LSHFT J  &mt LCTRL K  &mt LALT L         &mt LGUI SINGLE_QUOTE  &trans
&trans  &kp Z       &mt RIGHT_ALT X  &kp C        &kp V         &kp B        &kp C_MUTE    &trans       &kp N            &kp M        &kp COMMA    &mt RIGHT_ALT DOT  &kp FSLH               &trans
                    &trans           &trans       &lt 3 ESCAPE  &lt 1 SPACE  &lt 2 TAB     &lt 5 ENTER  &lt 4 BACKSPACE  &lt 6 DEL    &trans       &tog 7
            >;

            sensor-bindings =
                <&inc_dec_kp C_VOLUME_UP C_VOLUME_DOWN &inc_dec_kp UP_ARROW DOWN_ARROW>;

            label = "base";
        };

        navigation {
            bindings = <
&none  &none     &none     &none      &none      &none                       &none          &none        &none          &none        &none       &none
&none  &none     &none     &none      &none      &none                       &kp K_REDO     &kp K_PASTE  &kp K_COPY     &kp K_CUT    &kp K_UNDO  &none
&none  &kp LGUI  &kp LALT  &kp LCTRL  &kp LSHFT  &none                       &kp CAPSLOCK   &kp LEFT     &kp DOWN       &kp UP       &kp RIGHT   &none
&none  &none     &none     &none      &none      &none  &trans    &none      &kp INSERT     &kp HOME     &kp PAGE_DOWN  &kp PAGE_UP  &kp END     &none
                 &none     &none      &none      &none  &none     &kp ENTER  &kp BACKSPACE  &kp DELETE   &none          &none
            >;

            label = "nav";
        };

        mouse {
            bindings = <
&none  &none     &none     &none      &none       &none                      &none       &none        &none       &none      &none       &none
&none  &none     &none     &none      &none       &none                      &kp K_REDO  &kp K_PASTE  &kp K_COPY  &kp K_CUT  &kp K_UNDO  &none
&none  &kp LGUI  &kp LALT  &kp LCTRL  &kp LSHIFT  &none                      &none       &none        &none       &none      &none       &none
&none  &none     &none     &none      &none       &none  &none    &none      &none       &none        &none       &none      &none       &none
                 &none     &none      &none       &none  &none    &mkp RCLK  &mkp LCLK   &mkp MCLK    &none       &none
            >;

            label = "mouse";
        };

        media {
            bindings = <
&none  &none     &none     &none      &none      &none                       &none             &none            &none              &none            &none                             &none
&none  &none     &none     &none      &none      &none                       &rgb_ug RGB_TOG   &rgb_ug RGB_EFF  &rgb_ug RGB_BRD    &rgb_ug RGB_BRI  &rgb_ug RGB_COLOR_HSB(107,37,36)  &none
&none  &kp LGUI  &kp LALT  &kp LCTRL  &kp LSHFT  &none                       &none             &kp C_PREVIOUS   &kp C_VOLUME_DOWN  &kp C_VOLUME_UP  &kp C_NEXT                        &none
&none  &none     &none     &none      &none      &none  &none    &none       &out OUT_TOG      &bt BT_SEL 0     &bt BT_SEL 1       &bt BT_SEL 2     &bt BT_SEL 3                      &none
                 &none     &none      &none      &none  &none    &kp C_STOP  &kp C_PLAY_PAUSE  &kp C_MUTE       &none              &none
            >;

            label = "med";
        };

        numbers {
            bindings = <
&none  &none             &none         &none         &none         &none                                  &none  &none      &none      &none     &none     &none
&none  &kp LEFT_BRACKET  &kp N7        &kp N8        &kp N9        &kp RIGHT_BRACKET                      &none  &none      &none      &none     &none     &none
&none  &kp SEMICOLON     &kp NUMBER_4  &kp N5        &kp NUMBER_6  &kp EQUAL                              &none  &kp LSHFT  &kp LCTRL  &kp LALT  &kp LGUI  &none
&none  &kp GRAVE         &kp NUMBER_1  &kp NUMBER_2  &kp NUMBER_3  &kp BACKSLASH      &none        &none  &none  &none      &none      &none     &none     &none
                         &none         &none         &kp PERIOD    &kp N0             &kp MINUS    &none  &none  &none      &none      &none
            >;

            label = "num";
        };

        symbols {
            bindings = <
&none  &none           &none            &none         &none                 &none                                           &none  &none      &none      &none     &none     &none
&none  &kp LEFT_BRACE  &kp AMPERSAND    &kp ASTERISK  &kp LEFT_PARENTHESIS  &kp RIGHT_BRACE                                 &none  &none      &none      &none     &none     &none
&none  &kp COLON       &kp DOLLAR       &kp PERCENT   &kp CARET             &kp PLUS                                        &none  &kp LSHFT  &kp LCTRL  &kp LALT  &kp LGUI  &none
&none  &kp TILDE       &kp EXCLAMATION  &kp AT_SIGN   &kp POUND             &kp PIPE               &none             &none  &none  &none      &none      &none     &none     &none
                       &none            &none         &kp LEFT_PARENTHESIS  &kp RIGHT_PARENTHESIS  &kp UNDERSCORE    &none  &none  &none      &none      &none
            >;

            label = "sym";
        };

        function {
            bindings = <
&none  &none    &none   &none   &none      &none                              &none  &none      &none      &none     &none     &none
&none  &kp F12  &kp F7  &kp F8  &kp F9     &kp PRINTSCREEN                    &none  &none      &none      &none     &none     &none
&none  &kp F11  &kp F4  &kp F5  &kp F6     &kp SCROLLLOCK                     &none  &kp LSHFT  &kp LCTRL  &kp LALT  &kp LGUI  &none
&none  &kp F10  &kp F1  &kp F2  &kp F3     &kp PAUSE_BREAK  &none      &none  &none  &none      &none      &none     &none     &none
                &none   &none   &kp K_APP  &kp SPACE        &kp TAB    &none  &none  &none      &none      &none
            >;

            label = "fun";
        };

        gaming {
            bindings = <
&kp GRAVE   &kp N1  &kp N2           &kp N3  &kp N4     &kp N5                               &kp N6           &kp N7     &kp N8     &kp N9      &kp N0    &trans
&kp TAB     &kp Q   &kp W            &kp E   &kp R      &kp T                                &kp Y            &kp U      &kp I      &kp O       &kp P     &trans
&kp LSHFT   &kp A   &kp S            &kp D   &kp F      &kp G                                &kp H            &kp J      &kp K      &kp L       &kp SQT   &trans
&kp ESCAPE  &kp Z   &mt RIGHT_ALT X  &kp C   &kp V      &kp B     &kp C_MUTE    &trans       &kp N            &kp M      &kp COMMA  &kp PERIOD  &kp FSLH  &trans
                    &trans           &trans  &kp LSHFT  &kp LALT  &kp LCTRL     &lt 5 ENTER  &lt 4 BACKSPACE  &lt 6 DEL  &trans     &tog 0
            >;

            sensor-bindings =
                <&inc_dec_kp C_VOLUME_UP C_VOLUME_DOWN &inc_dec_kp UP_ARROW DOWN_ARROW>;

            label = "gaming";
        };
    };
};

```

haShinui, 2024-04-07:
Hello, I got the same thing, how did u fix it?

nickcoutsos, 2024-04-07:
That line references mouse behaviors but the required header is missing. Can you point me to your repo?

AhmedAA, 2024-04-07:
> That line references mouse behaviors but the required header is missing. Can you point me to your repo?

Thanks for answering.

I was missing the include and config lines documented here: <https://zmk.dev/docs/behaviors/mouse-emulation>.

I added them, but still get the error.

My repo is private, but will make it public momentarily.

AhmedAA, 2024-04-07:
Apparently can't make a private repo public. Or I don't know how to, I primarily use GitLab.

My sofle.conf looks like this:

```# Copyright (c) 2020 Ryan Cross
# SPDX-License-Identifier: MIT

# Uncomment the following line to enable the Sofle OLED Display
# CONFIG_ZMK_DISPLAY=y

# Uncomment these two lines to add support for encoders
CONFIG_EC11=y
CONFIG_EC11_TRIGGER_GLOBAL_THREAD=y

# Uncomment this line below to add rgb underglow / backlight support
CONFIG_ZMK_RGB_UNDERGLOW=y

CONFIG_ZMK_MOUSE=y

# Uncomment the line below to disable external power toggling by the underglow.
# By default toggling the underglow on and off also toggles external power
# on and off. This also causes the display to turn off.
# CONFIG_ZMK_RGB_UNDERGLOW_EXT_POWER=n
```

And my keymap looks the same as above, but I added the mouse header with the rest of the includes:
`#include <dt-bindings/zmk/mouse.h>`.

If this isn't enough, I'll just add you as a collaborator on the repo.

AhmedAA, 2024-04-07:
Found the problem.

I was missing another header:
`#include <dt-bindings/zmk/outputs.h>`.

It compiled now, and everything is in order.

Thanks for pointing in the right direction.

I dunno if the headers should be added to the generated keymap file. But hey, you also state that an understanding of zmk is still required when using this tool.

I will close the issue.

closed 2024-04-07 by AhmedAA

nickcoutsos, 2024-04-07:
When I added the mouse behavior it was still based on the experimental forks so I neglected to add the header to be automatically injected like other behaviors -- thanks for bringing this to my attention!

I've updated this now but I'm having problems deploying changes with my hosting platform at the moment, so if you're dependent on the GitHub integration to you'll still encounter this issue for the time being.

nickcoutsos, 2024-04-09:
I've migrated to a new platform so the backend is now running the latest which should include headers for the `&mkp` behavior.

AhmedAA, 2024-04-11:
> I've migrated to a new platform so the backend is now running the latest which should include headers for the `&mkp` behavior.

I can confirm that it added the mouse and output headers to my config now.

Again, thanks a million times for this tool and your dedication :)

radsonpatrick, 2024-05-03:
i have this problem ! my keymap editor not add 

#include <dt-bindings/zmk/outputs.h>

my repo https://github.com/radsonpatrick/zmk-config-corne/tree/test-keymap

nickcoutsos, 2024-05-03:
Hi @radsonpatrick, it looks like it added the header (in a weird spot, for sure, but that doesn't seem to be causing syntax problems). From the build logs it looks like the problem is coming from one of the mouse behaviors:

```
devicetree error: /__w/zmk-config-corne/zmk-config-corne/config/corne.keymap:118 (column 34): parse error: expected number or parenthesized expression
```

https://github.com/radsonpatrick/zmk-config-corne/blob/9eb75a413e65cf353b56029073e5df2adefa9d0d/config/corne.keymap#L118

Is it possible that you're building from a branch that doesn't have those specific mouse behavior features?

radsonpatrick, 2024-05-04:
you have a branch with mouse its work with keymap editor? 
i try with zmk original branch  and the branch  https://github.com/urob . but i do not have sucess using keymap editor but in other  project it's wotking editing manual, I found the difference between 

working other me project :
```
RAISE {
           bindings = <
   &kp N1 &kp N2 &kp N3 &kp N4 &kp N5   &kp N6 &kp N7  &kp N8     &kp N9   &kp N0
   &none &mkp LCLK &mmv MOVE_UP &mkp RCLK &msc SCRL_UP            &none &kp HOME  &kp UP &kp END   &kp PG_UP
   &none &mmv MOVE_LEFT &mmv MOVE_DOWN &mmv MOVE_RIGHT &msc SCRL_DOWN          &none &kp LEFT  &kp DOWN &kp RIGHT &kp PG_DN
        &mo ADVANCED &mt SPACE LCTRL &kp LGUI        &kp RET  &mt BSPC LSHIFT   &none
                        >;
                };
```
in your generate :

```
 nav_layer {
            label = "NAV";
            bindings = <
  &kp N1  &kp N2          &kp N3          &kp N4           &kp N5              &kp N6  &kp N7    &kp N8    &kp N9     &kp N0
  &none   &mkp LCLK       &mmv MOVE_UP    &mkp RCLK        &mwh SCROLL_UP      &none   &kp HOME  &kp UP    &kp END    &kp PG_UP
  &none   &mmv MOVE_LEFT  &mmv MOVE_DOWN  &mmv MOVE_RIGHT  &mwh SCROLL_DOWN    &teste  &kp LEFT  &kp DOWN  &kp RIGHT  &kp PG_DN
                          &mo 3           &trans           &trans              &trans  &trans    &trans
            >;
        };

```

the diference is in `&msc` and `&mwh`

the problem solution if you have branch with mouse is working ! kkkk 

thanks for your excelent job