nickcoutsos/keymap-editor #212: Profiles limited to 5 despite being set higher in config

quintuple-lained, 2024-04-10:
i set both `CONFIG_BT_MAX_CONN=` and `CONFIG_BT_MAX_PAIRED=` to 11 in my `.conf` file but the editor still only shows 5 profiles, is this a bug or am i misunderstanding the zmk config or is this unsupported?
[image]

nickcoutsos, 2024-04-10:
That's a limitation of the app, yeah.

I really only read the `.keymap` file or, mostly for layout generation purposes, the board's `.dts`/`.dtsi`/`.overlay`, there are so many config files that finding and checking them for relevant details on every load would introduce a ton of overhead.

Your best course of action right now would be, although admittedly very inconvenient, to manually adjust profile parameters after the fact. At that point the keymap editor would likely show an error for that binding but as long as you don't try to make any changes to that in the app it should be otherwise unaffected.

In the future... I guess the real solution is getting the available options from the keyboard itself when the protocol exists for it. Short of that, maybe I'll eventually have a "raw" mode where you can take responsibility for the contents of a key binding's parameters.

quintuple-lained, 2024-04-10:
i tried it and your app handles it without an error, it doesnt allow me to set the profile number to something higher than 4 but otherwise it just works
[image]
[image]