nickcoutsos/keymap-editor #213: Support for defines in config

ThomasAlban, 2024-04-13:
In my config I have defined `SUPER` as `LS(LA(LC(LCMD)))`. However when I use it as a keypress, it shows up in the editor as undefined. It would be nice if there was some way that this could be supported!

nickcoutsos, 2024-04-14:
Hi @ThomasAlban it's unfortunately not that straightforward to support. It may sound like there isn't a lot to it in this case but when opening the door to parsing preprocessor statements it's more a question of how complex it _could_ be. I wrote a little more about the complexity in [Keymap Editor Wiki::Preprocessor Support](https://github.com/nickcoutsos/keymap-editor/wiki/Preprocessor-Support)

ThomasAlban, 2024-04-14:
Yeah you're right - I was invisioning some sort of simple pre-processor but your writeup does a good job of explaining why it's not really possible to implement it simply!

ThomasAlban, 2024-04-14:
I just had a different idea about this which would perhaps make more sense - having the UI show `LS(LA(LC(LCMD)))` (in any order) as 'Hyper' and `LS(LA(LCTRL))` as 'Meh' (or maybe as symbols), perhaps as a togglable option.

nickcoutsos, 2025-05-22:
Yeah, that's fair. If either of you could confirm that the meh combo is always ctrl+alt+shift and has a commonly used symbol (ideally with a unicode glyph!) that'd be a great help! I took a cursory look into it before but only found anything definitive about hyper.