nickcoutsos/keymap-editor #216: Publish keymap page to Github

sebmos, 2024-04-20:
One thing I find quite useful about ZSA's Oryx is the ability to point someone to a link that shows my key map.

I was wondering about ways to provide that behaviour with the Keymap Editor. It could be via a page that includes the repo link (but without requiring Github permissions), that shows a stripped-down version without editing features. Alternatively, a preview could be published to the Github repo directly. I could see two ways for how that could work:

* Creating images of each layer, and just putting them in a README.md file into the repo (possibly in a subfolder to reduce the risk of collisions, but I'd actually prefer it in the root README.md).
* Creating actual HTML/JS - a subset of the keymap editor without any editing features. That would be the better experience for users, but of course would require setting up Github Pages.

nickcoutsos, 2024-04-30:
Hey, thanks for sharing this! I've been away so apologies for letting this hang without any response.

I like this idea for sure. I think it relates to the idea of a [keymap export](https://github.com/nickcoutsos/keymap-editor/discussions/141) type feature in a lot of basic ways but having the "rich" format HTML output would be really valuable. Likewise, being able to share diagrams via URL is a big deal.

Both of these are achievable to some extent, I think, with [keymap-drawer](https://github.com/caksoylar/keymap-drawer/). It's possible to incorporate it in a GitHub action so that static diagrams are rendered when the branch is updated and I suspect that you can specify SVG output to get tooltips but don't quote me on that...

Otherwise, yeah, I think I'd need a read-only version of keymap-editor's rendering components. Depending on how much interactivity is expected that may not be too crazy as, technically, I'm already generating HTML in some way, it's just a question of spitting it (and dependent JS/CSS) out. I think it'd be limited to static content, though, as I try to keep backend usage of the system to a minimum.

I'll probably keep this stewing in the back of my head.

sebmos, 2024-05-02:
I didn’t know about [keymap-drawer](https://github.com/caksoylar/keymap-drawer/) until after I posted this. It is a decent substitute and I’ve set it up for my repo.

It makes me wonder if the keymap-editor could integrate more tightly with it, eg providing the ability to set labels, icons etc for keyboard shortcuts and macros that would go into the keymap-drawer config file. 🤔