nickcoutsos/keymap-editor #220: Support additional OSX icons

mattrayner, 2024-05-03:
Hi, it would be cool if we could show OSX icons for additional keys:

- Brightness Up (pause break)
- Brightness Down (scroll lock)

Im happy to contribute a PR for this but have seen the closed source change you are making...

nickcoutsos, 2024-05-03:
Good call, I should include the brightness keys. I'm using FontAwesome currently and taking a quick peek at the collection it seems that brightness up/down aren't included in the free set. I've had other such limitations and have been meaning to explore other icon sets so this is just more justification for that.

Could you clarify what you mean with the pause break and scroll lock in your parentheses? I'm looking at the [ZMK Keycodes: Display](https://zmk.dev/docs/codes#display) table which just references HID codes. I'm on my MacBook where brightness are dedicated keys (touchbar, actually, bleh), are there fullsize external keyboards where those functions share the same keys?

mattrayner, 2024-05-06:
For OSX you typically use the keys SLCK for brightness down and PUSE_BREAK for brightness up as those keys are mapped by default at the OS level to brightness up and down

mattrayner, 2024-05-06:
For example, C_NEXT and C_PREV would be good to have an icon for:

[image]

mattrayner, 2024-05-06:
Backlight controls could too:

[image]

Along with the Bluetooth control icons

Have you looked at material design icons (MDI) [Material Design Icons](https://pictogrammers.com/library/mdi/)