nickcoutsos/keymap-editor #221: Feature request: quick swap

jacobrreed, 2024-05-11:
Ability to enter quick swap mode where you can click a button, and the next key you select will swap those two keys and so on until you exit quick swap mode
This is implemented in Oryx for moonlander if you want an idea of what I mean.

jacobrreed, 2024-05-11:
I can take a look at implementing this in a fork and see what happens

nickcoutsos, 2024-05-12:
Hmm... there's been a request already for generally being able to move existing bindings around (not sure when I'll be able to get to it though) so this is worth considering.

I do like the idea of it being a specific mode so that I don't have to worry about working in a click to select versus the existing click to open the binding dialog.

ArianVik-Art, 2025-06-28:
I think I have a similar request but related to the "quick assign" window: It would be nice if we could freely enter quick assign even after closing it the first time. 

As of now, when assigning a set of buttons and hitting "okay", the quick assign option disappears for that layer, and the only way to get it back seems to be to make a new layer.

nickcoutsos, 2025-06-29:
Quick assign is accessible via the three-dot menu next to the layer's name. The visual cue will also appear for layers with some sufficiently low portion of keys bound.

ArianVik-Art, 2025-06-29:
Oh, thank you! That's a relief

closed 2026-01-08 by neonvoidx